OpenXiangShan/XiangShan #2137: debug

kl123s, 2023-06-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

huxuan0307, 2023-06-15:
dmode在非debug模式不可写，保证了不接外部调试器的时候trigger都不会进入debug模式。hart处在debug模式必须有外部调试器配合。在m态通过指令写tdata1.dmode=1，再把hart交给外部调试器，在没有外部调试器的时候hart会表现为卡死。这样，任何m态程序都有主动让hart卡死的能力。

通过trigger进入Debug模式正确的用法是，
1. 连接外部调试器，让hart通过debug中断进入debug模式；
2. 设置trigger并把dmode置1；
3. 触发trigger并在debug模式响应。

Jeacy, 2023-10-17:
> dmode在非debug模式不可写，保证了不接外部调试器的时候trigger都不会进入debug模式。hart处在debug模式必须有外部调试器配合。在m态通过指令写tdata1.dmode=1，再把hart交给外部调试器，在没有外部调试器的时候hart会表现为卡死。这样，任何m态程序都有主动让hart卡死的能力。
> 
> 通过trigger进入Debug模式正确的用法是，
> 
> 1. 连接外部调试器，让hart通过debug中断进入debug模式；
> 2. 设置trigger并把dmode置1；
> 3. 触发trigger并在debug模式响应。

请问一下怎么获取program buffer的物理地址呀

wissygh, 2024-01-31:
> > dmode在非debug模式不可写，保证了不接外部调试器的时候trigger都不会进入debug模式。hart处在debug模式必须有外部调试器配合。在m态通过指令写tdata1.dmode=1，再把hart交给外部调试器，在没有外部调试器的时候hart会表现为卡死。这样，任何m态程序都有主动让hart卡死的能力。
> > 通过trigger进入Debug模式正确的用法是，
> > 
> > 1. 连接外部调试器，让hart通过debug中断进入debug模式；
> > 2. 设置trigger并把dmode置1；
> > 3. 触发trigger并在debug模式响应。
> > 
> > [TRANSLATION]
> > dmode is not writable in non-debug mode, which ensures that trigger will never go into debug mode without an external debugger. hart must be in debug mode with an external debugger. In the m mode, write tdata1.dmode=1 through the instruction, and then hand the hart to the external debugger. In this way, any M-state program has the ability to actively let hart get stuck.
> > The correct way to go into Debug mode via trigger is,
> > 
> > 1. Connect the external debugger and let hart enter debug mode through debug interrupt;
> > 2. Set trigger and set dmode to 1.
> > 3. trigger trigger and respond in debug mode.
> 
> 请问一下怎么获取program buffer的物理地址呀
> 
> [Translation] How do I get the physical address of the program buffer?

香山debug rom地址空间的基地址是0x38020000，progbuf地址取决于progbuf的大小，当前香山的progbuf地址空间为(0x38020380- 4*numProgbuf)~0x38020380

closed 2024-04-24 by wissygh