OpenXiangShan/XiangShan #2175: VCS仿真出错，是源代码的问题还是操作的问题？

JackyWu526, 2023-07-12:
**Describe the bug**
1) 执行make simv RELEASE=1
在有parameter string site;语句的地方，VCS仿真都会报错。需要手动改为 parameter string site=“xx”；赋值未某个字符才行。
2)执行./difftest/simv +workload=xxx 会报DPI某个单元找不到，例如FetchToIBuffer0_write，实际搜索也没找到对应的C文件，手动注释这些模块才行。

**To Reproduce**

**Expected behavior**

**Screenshots**

**Environment (optional, if necessary):**
 - OS: Ubuntu 20.04 LTS
 - Compiler: GCC 9.4.0
 - VCS : 2018

**Additional context**

wakafa1, 2023-07-12:
与 #2163 类似，可以尝试手动修改 DebugOptions 的选项为：
AlwaysBasicDiff=false
AlwaysBasicDB=false
我们之后会计划把这一步整合到 Makefile 中

JackyWu526, 2023-07-13:
请问是/src/main/scala/xiangshan/Parameters.scala这个文件里定义吧，只找到AlwaysBasicDiff，没有找到AlwaysBasicDB
commit： d156247e8cd8c1abb7477c72bd7874f8c0CC591

wakafa1, 2023-07-13:
好的，那只需要改变 AlwaysBasicDiff；AlwaysBasicDB 是最新的一些修改引入的选项

closed 2023-07-15 by wakafa1

JackyWu526, 2023-07-17:
你好，针对这个问题点， 如下改动，AlwaysBasicDiff = false后，make clean  然后make simv RELEASE=1，依然报一样的错误信息。
[image]

iMustard, 2023-08-18:
遇到相同的两个问题，按照 https://github.com/OpenXiangShan/XiangShan/issues/2163 修改后 make simv 没有解决