OpenXiangShan/XiangShan #2178: vcs 仿真报错

swanger, 2023-07-12:
```bash
 ./simv

Note-[RT_BS] Larger stack needed
  Note: Bumping stack limit from 8192 to 8220 Kbytes.

Chronologic VCS simulator copyright 1991-2016
Contains Synopsys proprietary information.
Compiler version L-2016.06_Full64; Runtime version L-2016.06_Full64;  Jul 12 17:25 2023
isWriteFetchToIBufferTable0 = 0
isWriteIfuWbToFtqTable0 = 0
isWriteFTQTable0 = 0
isWriteL2TlbPrefetchTable0 = 0
isWriteL1TlbTable0 = 0
isWritePageCacheTable0 = 0
isWritePTWTable0 = 0
isWriteL2TlbMissQueueTable0 = 0
isWriteBankConflictTable0 = 0
isWriteL1MissQMissTable0 = 0
isWriteLoadMissTable0 = 0
isFirstHitWrite0 = 0
ColdDownThreshold_0 = 0
isWriteInstInfoTable0 = 0
simv compiled at Jul 12 2023, 14:26:29
The image is ram.bin
Using simulated 8192MB RAM
Using simulated 32768B flash
[warning]no valid flash bin path, use preset flash instead
--diff is not given, try to use $(NEMU_HOME)/build/riscv64-nemu-interpreter-so by default
NemuProxy using /home/wanghairun/Projects/xs-env/NEMU/build/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled. 
[NEMU] Can not find flash image: (null)
[NEMU] Use built-in image instead
[src/device/io/mmio.c:38,add_mmio_map] Add mmio map 'flash' at [0x0000000010000000, 0x00000000100fffff]
No instruction of core 0 commits for 15000 cycles, maybe get stuck
(please also check whether a fence.i instruction requires more than 15000 cycles to flush the icache)
Let REF run one more instruction.
sh: spike-dasm: command not found

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 00800026a8 cmtcnt 2
commit group [01]: pc 0080002612 cmtcnt 1
commit group [02]: pc 0080002616 cmtcnt 1
commit group [03]: pc 008000261a cmtcnt 2
commit group [04]: pc 0080002622 cmtcnt 2
commit group [05]: pc 008000233a cmtcnt 1
commit group [06]: pc 00800026aa cmtcnt 1 <--
commit group [07]: pc 0080001d22 cmtcnt 2
commit group [08]: pc 0080001d26 cmtcnt 1
commit group [09]: pc 008000232e cmtcnt 1
commit group [10]: pc 0080002330 cmtcnt 2
commit group [11]: pc 008000268a cmtcnt 1
commit group [12]: pc 008000268e cmtcnt 1
commit group [13]: pc 0080002690 cmtcnt 2
commit group [14]: pc 0080002698 cmtcnt 2
commit group [15]: pc 00800026a0 cmtcnt 2

============== Commit Instr Trace ==============
commit inst [00]: pc 0080001d1e inst 0b913423 wen 0 dst 00000008 data 00000000800026f0 robidx 000000 sqidx 000000
commit inst [01]: pc 0080001d20 inst 0ba13023 wen 0 dst 00000000 data 00000000800026f0 robidx 000001 sqidx 000001
commit inst [02]: pc 0080001d22 inst 09b13c23 wen 0 dst 00000018 data 00000000800026f0 robidx 000002 sqidx 000002
commit inst [03]: pc 0080001d24 inst 00a12e23 wen 0 dst 0000001c data 00000000800026f0 robidx 000003 sqidx 000003
commit inst [04]: pc 0080001d26 inst 608000ef wen 1 dst 00000001 data 0000000080001d2a robidx 000004             
commit inst [05]: pc 008000232e inst ff010113 wen 1 dst 00000002 data 000000008000cee0 robidx 000005             
commit inst [06]: pc 0080002330 inst 00113423 wen 0 dst 00000008 data 00000000800026f0 robidx 000006 sqidx 000004
commit inst [07]: pc 0080002332 inst 358000ef wen 1 dst 00000001 data 0000000080002336 robidx 000007             
commit inst [08]: pc 008000268a inst 400017b7 wen 1 dst 0000000f data 0000000040001000 robidx 000008             
commit inst [09]: pc 008000268e inst 0007a783 wen 1 dst 0000000f data 0000000000000000 robidx 000009 (skip) lqidx 000000
commit inst [10]: pc 0080002690 inst 0107d71b wen 1 dst 0000000e data 0000000000000000 robidx 00000a             
commit inst [11]: pc 0080002694 inst 00002697 wen 1 dst 0000000d data 0000000080004694 robidx 00000b             
[... 40 lines cut ...]
satp: 0x0000000000000000
mip: 0x0000000000000000 mie: 0x0000000000000000 mscratch: 0x0000000000000000 sscratch: 0x0000000000000000
mideleg: 0x0000000000000000 medeleg: 0x0000000000000000
mtval: 0x0000000000000000 stval: 0xac28c658ac28c658 mtvec: 0x0000000000000000 stvec: 0x0000000000000000
privilege mode:3  pmp: below
 0: cfg:0x00 addr:0x0000000000000000| 1: cfg:0x00 addr:0x0000000000000000
 2: cfg:0x00 addr:0x0000000000000000| 3: cfg:0x00 addr:0x0000000000000000
 4: cfg:0x00 addr:0x0000000000000000| 5: cfg:0x00 addr:0x0000000000000000
 6: cfg:0x00 addr:0x0000000000000000| 7: cfg:0x00 addr:0x0000000000000000
 8: cfg:0x00 addr:0x0000000000000000| 9: cfg:0x00 addr:0x0000000000000000
10: cfg:0x00 addr:0x0000000000000000|11: cfg:0x00 addr:0x0000000000000000
12: cfg:0x00 addr:0x0000000000000000|13: cfg:0x00 addr:0x0000000000000000
14: cfg:0x00 addr:0x0000000000000000|15: cfg:0x00 addr:0x0000000000000000
priviledgeMode: 3
$finish called from file "/home/wanghairun/Projects/xs-env/XiangShan/difftest/src/test/vsrc/vcs/top.v", line 152.
$finish at simulation time                33235
           V C S   S i m u l a t i o n   R e p o r t 
Time: 33235 ns
CPU Time:     41.830 seconds;       Data structure size:  66.5Mb

```

请问一下，这个vcs仿真是已经跑起来了吗？为什么用的默认的ready-to-run目录下的coremark case会报错？

谢谢！

wakafa1, 2023-07-13:
请提供一下 VCS 仿真的香山版本，我们这边复现检查一下

swanger, 2023-07-14:
git submodule
ca590e7bf514f74ce0a841e91c497d0b7734ddd1 coupledL2 (pass-dualcore-tl-test-92-gca590e7)
 0a1e5e29919cac3fd60d3f51620426645c01710b difftest (heads/master)
 43474be4c76bb0afe3b2d8358e26f3c37c38b380 fudian (remotes/origin/fadd-adjust-7-g43474be)
 b7308d958dfa7073e47ca10ce3974d267592049c huancun (remotes/origin/nanny-9-gb7308d9)
 ff39f3680f69f9b38fa3cdf4c9fc1b2c6b3ec919 ready-to-run (remotes/origin/nanhu-pmp-fix-2-gff39f36)
 254ebf7150e1d633fee123281f723d9936d30075 rocket-chip (remotes/origin/xs-merge-41-g254ebf715)
 3d812fec9936ccd584df7721aa8c2d02e932d325 utility (remotes/origin/fix-chiseldb)

git log
commit a483ee06bac87d8420ac99e1f554911d7d3d8688 (HEAD -> master, origin/master, origin/HEAD)
Author: Guokai Chen <chenguokai17@mails.ucas.ac.cn>
Date:   Thu Jul 6 13:08:00 2023 +0800

    Predecode: fix unintended width cast (#2150)

...

wakafa1, 2023-07-17:
我们按照你提供的代码版本运行了 vcs 仿真，无法复现问题，请再仔细检查运行流程。可以贴一下你执行的所有命令

maolei-l, 2023-07-17:
@wakafa1 
> sh: spike-dasm: command not found
log文件中，为何会出现该该命令未能找到？

wakafa1, 2023-07-17:
这是 Spike 项目提供的一个机器码转汇编的工具，可以参考 https://github.com/riscv-software-src/riscv-isa-sim
如果没有安装也不会影响仿真结果

maolei-l, 2023-07-17:
好的。
XiangShan仓库里面只有Verilator以及VCS相关的Makefile文件。
Xiangshan是否只支持vcs而不支持xrun？

wakafa1, 2023-07-17:
香山目前支持 Verilator 和 VCS，暂不支持 xrun

maolei-l, 2023-07-17:
目前我正在使用xrun运行XiangShan项目，但是中间出现了一些问题，其中一个和该issue一样。

能否告知一下，你们项目组不支持xrun的理由吗？
是由于xrun的特性不支持XiangShan项目；还是由于你们并没有去开发相关的xrun mk。
如果是后者，后续我本地跑通后，可以将相关的mk上传到该仓库。

wakafa1, 2023-07-17:
我们目前暂时没有做 xrun 的适配，欢迎你给香山贡献代码

ppx-bit, 2023-07-31:
@swanger 你好，请问这个问题后续你是怎么解决的？

closed 2025-09-05 by Yan-Muzi