OpenXiangShan/XiangShan #2190: Modify Multiplier.scala to generate pure combination multiplier instead of a time series multiplier.

whitecloudsun, 2023-07-19:
I would like to ask if the multiplier of the Xiangshan integer module can be achieved by modifying the Chisel configuration to generate a pure combination multiplier design.

 I only modified the class ArrayMultiplier function in Multiplier.scala fiel into following two versions, and when I execute the make verilog command in the end, errors will be reported.

//first modification
class ArrayMultiplier(len: Int)(implicit p: Parameters)
  extends AbstractMultiplier(len) {

  override def latency = 2

  val mulDataModule = Module(new ArrayMulDataModule(len))
  mulDataModule.io.a := io.in.bits.src(0)
  mulDataModule.io.b := io.in.bits.src(1)
  mulDataModule.io.regEnables := VecInit((1 to latency) map (i => regEnable(i)))
  
  mulDataModule.io.result := mulDataModule.io.a * mulDataModule.io.b
  val result = mulDataModule.io.result

  io.out.bits.data := result

  XSDebug(p"validVec:${Binary(Cat(validVec))} flushVec:${Binary(Cat(flushVec))}\n")
}

//second modification
class ArrayMultiplier(len: Int)(implicit p: Parameters)
  extends AbstractMultiplier(len) {

  val mulDataModule = Module(new ArrayMulDataModule(len))
  
  val result = io.in.bits.src(0) * io.in.bits.src(1)

  io.out.bits.data := result

  //XSDebug(p"validVec:${Binary(Cat(validVec))} flushVec:${Binary(Cat(flushVec))}\n")
}

 I would like to ask if there are any issues or incomplete modifications made by me, or is it not possible to convert the configuration to generate a pure combination multiplier design at all. Looking forward to your reply. Thank you very much!

Tang-Haojin, 2023-07-19:
For your first modification, `mulDataModule.io.regEnables` is an output port and you may not assign it with any other signal outside the `mulDataModule` itself.

For your second modification, the IO ports of `mulDataModule` are not assigned.

If you want to generate a pure combination multiplier, no trait `HasPipelineReg` or module `ArrayMulDataModule` is needed. You may simply extend a new subclass of `AbstractMultiplier`, say, `CombinationMultiplier`, then your code may be like:

``` scala
class CombinationMultiplier(len: Int)(implicit p: Parameters) extends AbstractMultiplier(len) {
  val (src1, src2) = (io.in.bits.src(0), io.in.bits.src(1))
  val mulRes = (src1.asSInt * src2.asSInt).asUInt
  val xlen = io.out.bits.data.getWidth
  val res = Mux(ctrl.isHi, mulRes(2*xlen-1, xlen), mulRes(xlen-1,0))

  io.out.bits.data := Mux(ctrl.isW, SignExt(res(31,0),xlen), res)
  io.out.bits.uop := io.in.bits.uop
  io.in.ready := true.B
  io.out.valid := io.in.fire
}
```

Do not forget to change `ArrayMultiplier` into your `CombinationMultiplier` in the other parts of the code.

closed 2023-07-21 by Tang-Haojin

whitecloudsun, 2023-07-21:
Thank you very much for your guidance. It has already worked！