OpenXiangShan/XiangShan #2212: FPGA更换型号时 XDMA

Lei-DaQQ, 2023-08-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior. Please include the commit hash and your modification as well.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (optional, if necessary):**
 - OS: [e.g. Ubuntu 20.04 LTS]
 - Compiler: [e.g. GCC 9.4.0]

**Additional context**
Add any other context about the problem here.

xianfeili, 2023-08-07:
xdma_0的block design在最小系统里是没有使用到的。可以在vivado里右键disable file。这样就不会再检查xdma的版本了

xianfeili, 2023-08-07:
> **Describe the bug** 部署在fpga上时，将 project device更换成自己的芯片型号，更新ip时有如下问题： [image]
> 
> ```
> * IP 'xdma_xdma_0_0' does not support the current project part 'xczu9eg-ffvb1156-2-e'. Note that part differences may result in undefined behavior. * IP 'xdma_xdma_0_0' contains one or more locked subcores.
> ```
> 
> 请教一下这个问题是板子不支持吗?
> 
> 版本是 vivado 20.2

xdma_0的block design在最小系统里是没有使用到的。可以在vivado里右键disable file。这样就不会再检查xdma的版本了

Lei-DaQQ, 2023-08-07:
好的，感谢

closed 2023-08-08 by Lei-DaQQ