OpenXiangShan/XiangShan #2214: 请问MinimalConfig配置有推荐的FPGA开发板型号吗？

erhade, 2023-08-03:


Lei-DaQQ, 2023-08-04:
占坑，(实验室的Zynq zcu102应该不行

wakafa1, 2023-08-07:
各个分支的 MinimalConfig 是我们用来做仿真冒烟测试和简单 Demo 的一个配置，没有经过认真的验证，我们不建议用它做 FPGA 的 prototype；
如果大家想在 FPGA 上跑一个稳定的裁剪版香山，我们推荐使用 nanhu-G 分支的 NanHuGServeConfig，这是我们做的一个较为平衡裁剪的版本，在 vcu128 开发版(vu37p芯片)下约占 70% 的 LUT 和 27% 的 BRAM

erhade, 2023-08-07:
@wakafa1 感谢！请问在vcu128 开发版运行nanhu-G 分支的 NanHuGServeConfig，能支持上层跑Linux内核吗？（之前看到了你们在VU19P上面双核运行Linux桌面）

wakafa1, 2023-08-07:
可以跑 Linux 内核、busybox 等 workload，但是 Linux 发行版没有跑过

erhade, 2023-08-07:
@wakafa1 我看到去年的issue，如果想跑完整版的nanhu，里面提到你们用的开发板是S2C的vu19p。能否告知你们现在使用的具体开发板型号？是单颗FPGA的P-LSVU19PS-2（S7-19PS-2逻辑系统）吗？感谢。

xianfeili, 2023-08-08:
> @wakafa1 我看到去年的issue，如果想跑完整版的nanhu，里面提到你们用的开发板是S2C的vu19p。能否告知你们现在使用的具体开发板型号，是单颗FPGA的P-LSVU19PS-2（S7-19PS-2逻辑系统）吗？使用其他的搭载vu19p的板子可以吗，比如说htg-960？

我们手头上是双路FPGA的Prodigy™ S7-19PD Logic System，最小系统只用了其中的一个FPGA。

xianfeili, 2023-08-09:
> @xianfeili 感谢您的解答，我还有几个问题想问： q1. 您说的最小系统指的是MinimalConfig吗？ q2. 我看到之前的issue里面说“默认配置的香山在vu19p上占用大概40-50%的lut资源”，请问是否是nanhu默认配置在双路FPGA上占用大概40-50%的lut资源？ q3. 如果我们想跑现在默认配置的nanhu（不是MinimalConfig），单路vu19p能跑起来吗？ q4. 如果未来发布香山第三代昆明湖，单路vu19p是否还够用？

1. 这里的最小系统指的是SOC的最小系统，CPU的配置是release的完整版配置
2. 香山是在单路的VU19P上做的验证，nanhu的默认配置是单颗VU19P的资源用量
3. 跑默认配置的话，单颗VU19P是足够的
4. 昆明湖目标是四核，目前预估是至少需要两颗VU19P才能跑完整版的昆明湖

Tang-Haojin, 2023-12-13:
我们应该已经解决了您的问题。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

Your issue may have been solved. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2023-12-13 by Tang-Haojin