OpenXiangShan/XiangShan #2257: make verilog -j9 NUM_CORES=2失败

JLY0226, 2023-08-28:
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at firrtl.ir.Block.mapStmt(IR.scala:653)
	at firrtl.Mappers$StmtMagnet$$anon$3.map(Mappers.scala:31)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:48)
	at firrtl.passes.RemoveCHIRRTL$.collect_refs(RemoveCHIRRTL.scala:208)
	at firrtl.passes.RemoveCHIRRTL$.$anonfun$remove_chirrtl_m$3(RemoveCHIRRTL.scala:315)
	at firrtl.passes.RemoveCHIRRTL$$$Lambda$41365/0x00000008443d0840.apply(Unknown Source)
	at firrtl.ir.Module.mapStmt(IR.scala:1198)
	at firrtl.Mappers$ModuleMagnet$$anon$14.map(Mappers.scala:108)
	at firrtl.Mappers$ModuleMap$.map$extension(Mappers.scala:121)
	at firrtl.passes.RemoveCHIRRTL$.remove_chirrtl_m(RemoveCHIRRTL.scala:315)
	at firrtl.passes.RemoveCHIRRTL$.$anonfun$execute$1(RemoveCHIRRTL.scala:323)
	at firrtl.passes.RemoveCHIRRTL$$$Lambda$41360/0x00000008443d3040.apply(Unknown Source)
	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:286)
	at scala.collection.TraversableLike$$Lambda$21/0x00000008400f0840.apply(Unknown Source)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at scala.collection.TraversableLike.map(TraversableLike.scala:286)
	at scala.collection.TraversableLike.map$(TraversableLike.scala:279)
	at scala.collection.AbstractTraversable.map(Traversable.scala:108)
	at firrtl.passes.RemoveCHIRRTL$.execute(RemoveCHIRRTL.scala:323)
	at firrtl.Transform.transform(Compiler.scala:280)
	at firrtl.Transform.transform$(Compiler.scala:280)
	at firrtl.passes.RemoveCHIRRTL$.transform(RemoveCHIRRTL.scala:18)
	at firrtl.stage.transforms.ExpandPrepares.execute(ExpandPrepares.scala:19)
	at firrtl.Transform.transform(Compiler.scala:280)
	at firrtl.Transform.transform$(Compiler.scala:280)
	at firrtl.stage.transforms.ExpandPrepares.transform(ExpandPrepares.scala:7)
	at firrtl.stage.transforms.CatchCustomTransformExceptions.execute(CatchCustomTransformExceptions.scala:10)
	at firrtl.Transform.transform(Compiler.scala:280)
	at firrtl.Transform.transform$(Compiler.scala:280)
	at firrtl.stage.transforms.CatchCustomTransformExceptions.transform(CatchCustomTransformExceptions.scala:7)
1 targets failed
XiangShan.runMain subprocess failed
make: *** [Makefile:73: build/XSTop.v] Error 1

已修改build.sc中Xmx64为Xmx16G，依旧报错

wakafa1, 2023-08-29:
16G 对于双核的编译还是小了一些，可以尝试在大内存机器上编译一下看是否出错

yaanng2022, 2023-11-21:
修改32G跑了几次也是一样的错误

chenguokai, 2023-11-21:
32G内存通常也不够。请考虑使用64G甚至128G内存机器或添加swap。
32G RAM is not sufficient. Please retry on machines with 64/128G RAM or enable swap.

Tang-Haojin, 2023-12-13:
我们应该已经解决了您的问题。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

Your issue may have been solved. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2023-12-13 by Tang-Haojin