OpenXiangShan/XiangShan #2305: 请教一个关于BankedDataArray里的read_result ecc问题

janboeye, 2023-09-15:
你好

在BankedDataArray的代码里，
```
read_result(bank_index)(way_index).error_delayed := dcacheParameters.dataCode.decode(ecc_data_delayed).error
```
我从波形上看，L1Dcache里error_delayed一直为高，不知道这个是否正常。

谢谢

linjuanZ, 2023-09-19:
您好，`read_result` 这个信号没有考虑使能信号，您还需要从波形里面关注一下读使能是否有效以及是否存在 bank 冲突：
```scala
(0 until LoadPipelineWidth).map(i => {
    // ... ...
    (0 until VLEN/DCacheSRAMRowBits).map( j =>{
      io.read_resp_delayed(i)(j) := bank_result_delayed(rr_div_addr)(rr_bank_addr(j))
      // error detection
      io.read_error_delayed(i)(j) := rr_read_fire && read_bank_error_delayed(rr_div_addr)(rr_bank_addr(j)) && !RegNext(io.bank_conflict_slow(i))
    })
  })

```

janboeye, 2023-09-19:
多谢。commit 62dfd6c3b0f051d061f4e8c2c855ef69ecba0d94上没有这处的改动，我再试试最新的代码。

closed 2023-09-19 by janboeye