OpenXiangShan/XiangShan #2313: Error when running Debian in NEMU

Livebewater, 2023-09-18:
```
[   28.930000] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[   28.930000] VFS: Mounted root (ext4 filesystem) readonly on device 179:1.
[   28.930000] Freeing unused kernel memory: 116K
[   28.930000] This architecture does not have kernel memory protection.
/bin/sh: 0: cannot open earlycon: No such file
[   37.320000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
[   37.320000]
[   37.320000] CPU: 0 PID: 1 Comm: sh Not tainted 4.18.0-14487-gbcdfa1f46021 #44
[   37.320000] Call Trace:
[   37.320000] [<ffffffe00001ee60>] walk_stackframe+0x0/0xa6
[   37.320000] [<ffffffe00001efd6>] show_stack+0x2a/0x34
[   37.320000] [<ffffffe0003306a0>] dump_stack+0x20/0x28
[   37.320000] [<ffffffe000022e2a>] panic+0xc0/0x1cc
[   37.320000] [<ffffffe000023fce>] do_exit+0x5ae/0x612
[   37.320000] [<ffffffe000024848>] do_group_exit+0x22/0x5e
[   37.320000] [<ffffffe00002489c>] __wake_up_parent+0x0/0x22
[   37.320000] [<ffffffe00001ddb4>] ret_from_syscall+0x0/0xe
[   37.320000] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
[   37.320000]  ]---
```

Livebewater, 2023-09-18:
when i change "earlycon" to "earlycon=sbi" in bootargs just like in the comment in nemu.dtsi (my debian kernel is 5.4.0-152), will have a new error and hang after i typing any command
```
[   28.260000] VFS: Mounted root (ext4 filesystem) readonly on device 179:1.
[   28.260000] Freeing unused kernel memory: 116K
[   28.260000] This architecture does not have kernel memory protection.
/bin/sh: 0: can't access tty; job control turned off
# ls

```

wakafa1, 2023-09-19:
仿真环境下，NEMU 不接收终端中的输入作为串口输入，因此您打印的 ls 是无效的。想要给仿真环境下的系统串口添加输入，请参考 https://github.com/OpenXiangShan/NEMU/blob/master/src/device/serial.c 中的 present_input 函数

Tang-Haojin, 2023-12-13:
我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2023-12-13 by Tang-Haojin