OpenXiangShan/XiangShan #2345: DiffTest如何支持SOC级验证，目前看香山重点利用DiffTest验证Core的功能，后续是否考虑拓展到整个SOC

changekkk, 2023-09-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

closed 2023-09-28 by poemonsense

changekkk, 2023-10-08:
目前香山的SOC验证，是只在FPGA端进行吗？仿真端如果验证的话，特别是使用商业VIP的话，对于SOC的验证策略是什么？

poemonsense, 2023-10-08:
香山是一个处理器核CPU CORE项目，我们只关注CPU对SOC的支持，SOC的验证不在香山项目的考虑范围之内。