OpenXiangShan/XiangShan #2349: Consider Providing an Alternative Project Host/Mirror

abebeos, 2023-09-28:
**Is your feature request related to a problem? Please describe.**

* One cannot trust "The West"
* Risk that this project looses its github repository

**Describe the solution you'd like**

* Stay on github, but provide an additional project-platform (mirror) that is independent of western control
* Advertise the mirror prominently within this repository

### Background

While I usually keep politiccs out of code-work, this project here interests me far too much, and I do not want to 'loose' it some day from my radar, because some western potiticians had a bad dream.

So, please, an independent chip-architecture needs independent project-hosts, too.

Tang-Haojin, 2023-10-07:
We have a gitee mirror at https://gitee.com/OpenXiangShan/XiangShan.

abebeos, 2023-10-07:
Very nice! Please consider mentioning this somewhere (e.g. in the README). Closing Issue as I'm covered!

closed 2023-10-07 by abebeos