OpenXiangShan/XiangShan #2363: 关于香山在windows环境下部署生成verilog失败的问题

Jeacy, 2023-10-09:
**Describe the bug**

感谢香山开源项目。我首先git clone了香山最新的代码并进行了初始化，然后我在github上下载了最新版本的的mill（0.11.5），使用./mill.bat -i mill.scalalib.GenIdea/idea命令生成索引成功，最后我想要生成verilog代码，我在window终端内使用.\mill.bat -i XiangShan.test.runMain top.SimTop -td build --config DefaultConfig --infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf --gen-mem-verilog full --num-cores 1 --enable-difftest命令，出现构建错误。经过排除应该不是内存大小问题和mill版本问题，想问下有遇到类似问题并且如何解决的吗？

**To Reproduce**
Steps to reproduce the behavior. Please include the commit hash and your modification as well.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Environment (optional, if necessary):**

**Additional context**
Add any other context about the problem here.

wakafa1, 2023-10-09:
您好，我们目前经过测试的开发环境为 Ubuntu 20.04 LTS，强烈建议您在该系统下进行开发

Jeacy, 2023-10-09:
好的，感谢您的回答

closed 2023-10-21 by cebarobot