OpenXiangShan/XiangShan #2392: 请教关于chisel3.0升级到chisel6.0 #2372

user-stu-l, 2023-10-18:
请问具体如何配置chisel6+CIRCT环境来编译生成verilog？chisel6的源代码在哪个文件夹？

Tang-Haojin, 2023-10-18:
在master分支下，你需要将 [firtool](https://github.com/llvm/circt/releases) 加入到环境变量，然后 `make emu MFC=1` 即可。

例如，如果你的设备是 64 位 x86 linux 系统，则需要下载 `firrtl-bin-linux-x64.tar.gz`，并在解压缩后将 `firtool-<version>/bin` 加入到系统的 `PATH` 环境变量中。

poemonsense, 2023-10-19:
@Tang-Haojin possibly we can update the readme.md

Tang-Haojin, 2023-10-19:
Make sence. Let me do it.

nlgzcloud, 2023-10-19:
将firtool加入到PATH环境变量后，执行：make verilog MFC=1，结果报错如下，请问是什么原因导致的，怎样解决？谢谢

Exception in thread "main" firrtl.options.OptionsException: Unknown option --disable-all
	at firrtl.options.ExceptOnError.reportError(OptionParser.scala:25)
	at firrtl.options.ExceptOnError.reportError$(OptionParser.scala:25)
	at firrtl.options.BareShell$$anon$1.reportError(Shell.scala:21)
	at scopt.OptionParser$$anon$4.reportError(OptionParser.scala:191)
	at scopt.ORunner$.$anonfun$runEffects$1(ORunner.scala:438)
	at scopt.ORunner$.$anonfun$runEffects$1$adapted(ORunner.scala:435)
	at scala.collection.immutable.List.foreach(List.scala:333)
	at scopt.ORunner$.runEffects(ORunner.scala:435)
	at scopt.OptionParser.parse(OptionParser.scala:188)
	at firrtl.options.BareShell$$anon$1.firrtl$options$DuplicateHandling$$super$parse(Shell.scala:21)
	at firrtl.options.DuplicateHandling.parse(OptionParser.scala:51)
	at firrtl.options.DuplicateHandling.parse$(OptionParser.scala:32)
	at firrtl.options.BareShell$$anon$1.parse(Shell.scala:21)
	at firrtl.options.BareShell.parse(Shell.scala:35)
	at firrtl.options.Stage.execute(Stage.scala:56)
	at top.Generator$.execute(Generator.scala:37)
	at top.TopMain$.delayedEndpoint$top$TopMain$1(Top.scala:255)
	at top.TopMain$delayedInit$body.apply(Top.scala:244)
	at scala.Function0.apply$mcV$sp(Function0.scala:42)
	at scala.Function0.apply$mcV$sp$(Function0.scala:42)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
	at scala.App.$anonfun$main$1(App.scala:98)
	at scala.App.$anonfun$main$1$adapted(App.scala:98)
	at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:575)
	at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:573)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:933)
	at scala.App.main(App.scala:98)
	at scala.App.main$(App.scala:96)
	at top.TopMain$.main(Top.scala:244)
	at top.TopMain.main(Top.scala)
1 targets failed
xiangshan[chisel].runMain subprocess failed

poemonsense, 2023-10-20:
Please try #2398

closed 2023-10-20 by poemonsense