OpenXiangShan/XiangShan #2396: 访存子系统中的异常处理

Byiza, 2023-10-19:
1. 请教一下load流水线中的异常具体是怎么处理的呢？返回ROB，等异常指令到达ROB队尾的时候再处理异常并清空load queue嘛？可以返回给load queue处理、清空异常load指令及之后的所有entry嘛？
2. 以及前递失败后为什么会触发重定向呢？实地址不一致的话不前递、通过load流水线向cache取数据不可以嘛？是会引发什么错误嘛？

以上两个问题在文档里写的比较散，可以帮忙总结解释一下嘛，谢谢！

happy-lx, 2023-10-21:
> 请教一下load流水线中的异常具体是怎么处理的呢？返回ROB，等异常指令到达ROB队尾的时候再处理异常并清空load queue嘛？可以返回给load queue处理、清空异常load指令及之后的所有entry嘛？
1. 如果 load 流水线中发生的是不可以通过重发解决的异常，例如 page fault，access fault，这种需要异常处理程序帮助处理，就需要返回给 ROB，等异常指令到达 ROB 队尾的时候再处理异常并清空 load queue。
2. 如果是可以通过重发解决的异常，例如 tlb miss，cache miss，是先返回到 LoadQueueReplay，等待回填后再重发到流水线。
3. 另外还有还有一种 store-load 违例的异常，load 没有前递到 store 的数据提前写回的情况，在被检测出来后使用类似分支预测错误恢复的方式解决，重定向信号返回给 load queue 处理、清空违例的 load 指令及之后的所有 entry。

> 以及前递失败后为什么会触发重定向呢？实地址不一致的话不前递、通过load流水线向cache取数据不可以嘛？是会引发什么错误嘛？
前递时虚地址一致，实地址不一致的情况很少，用处理异常的方式触发重定向来解决比较简单。这种情况下可能会读错数据，sbuffer 里面如果有两项有相同虚地址，不同实地址，前递到的数据是错的，检测到这种情况会触发重定向，把已经提交掉的 store 刷到 cache 里面。

Tang-Haojin, 2023-12-13:
我们应该已经解决了您的问题。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

Your issue may have been solved. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2023-12-13 by Tang-Haojin