OpenXiangShan/XiangShan #2403: make riscv-pk ERROR: ../machine/dtb.S:5: Error: file not found: system.dtb

fzhang1991, 2023-10-20:
**Describe the bug**
# make riscv-pk get ERROR: ../machine/dtb.S:5: Error: file not found: system.dtb
**To Reproduce**
cd riscv-pk
mkdir build
cd build
../configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gc_zifencei
make

then get Error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

wakafa1, 2023-10-21:
请根据我们提供文档的步骤来使用 riscv-pk，详见 https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/linux-kernel-for-xs/

Please follow the steps to use riscv-pk as documented at https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/linux-kernel-for-xs-en/

Tang-Haojin, 2023-12-13:
我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2023-12-13 by Tang-Haojin