OpenXiangShan/XiangShan #2417: Missing header files in constantin.cpp

Emin-stack, 2023-10-24:
**Describe the bug**
I'm using the *master branch* to run emu with Verilator. Every time when I executing `make emu CONFIG=MinimalConfig EMU_THREADS=4 -j16`, compiler report missing header files error.

```
Tue, 24 Oct 2023 10:06:20 +0800
time -a -o /home/qi/Workbench/XiangShan/build/time.log make -s VM_PARALLEL_BUILDS=1 OPT_FAST="-O3" -C /home/qi/Workbench/XiangShan/build/emu-compile -f VSimTop.mk
/home/qi/Workbench/XiangShan/build/constantin.cpp:6:13: error: ‘uint64_t’ was not declared in this scope
    6 | map<string, uint64_t> constantinMap;
      |             ^~~~~~~~
/home/qi/Workbench/XiangShan/build/constantin.cpp:4:1: note: ‘uint64_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
    3 | #include <string>
  +++ |+#include <cstdint>
    4 | using namespace std;
/home/qi/Workbench/XiangShan/build/constantin.cpp:6:21: error: template argument 2 is invalid
    6 | map<string, uint64_t> constantinMap;
      |                     ^
/home/qi/Workbench/XiangShan/build/constantin.cpp:6:21: error: template argument 4 is invalid
/home/qi/Workbench/XiangShan/build/constantin.cpp: In function ‘void constantinInit()’:
/home/qi/Workbench/XiangShan/build/constantin.cpp:9:43: error: assignment of read-only location ‘"StoreBufferThreshold_0"[constantinMap]’
    9 |   constantinMap["StoreBufferThreshold_0"] = 7;
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

...

```

**To Reproduce**
1. `git checkout master`
2. `make emu CONFIG=MinimalConfig EMU_THREADS=2 -j16`
3. Then compiler report this error. (And It always happens.)

**Expected behavior**
The missing header files should be added in object Constantin(utility/src/main/scala/utility/Constantin.scala)
```
--- a/utility/src/main/scala/utility/Constantin.scala
+++ b/utility/src/main/scala/utility/Constantin.scala
@@ -120,6 +120,7 @@ object Constantin extends ConstantinParams {
      s"""
         |#include <map>
         |#include <string>
+        |#include <stdint.h>
         |using namespace std;
         |
         |map<string, uint64_t> constantinMap;
```
Then it works well and can be run with Verilator emu :)

**Environment :**
 - OS: Fedora Linux 38 (Workstation Edition) x86_64
 - Compiler: gcc version 13.2.1 20231011 (Red Hat 13.2.1-4) (GCC)
 - mill : Mill Build Tool version 0.11.1
 - java : Java version: 17.0.8, vendor: Red Hat, Inc., runtime: /usr/lib/jvm/java-17-openjdk-17.0.8.0.7-1.fc38.x86_64
 - Verilator : Verilator 5.016 2023-09-16 rev v5.014-149-g57c816f90

**Additional context**
Or maybe the problem is an environmental issue that only happens on fedora and not on ubuntu(Please ignore me)

wakafa1, 2023-10-25:
你提出的问题我们在ubuntu上没有复现，但看起来是有道理的，我们后续会考虑加上这个头文件，感谢你的报告

We have not reproduced the problem you raised on Ubuntu, but the idea seems reasonable. We will consider adding this header file in the future. Thank you for your report.

cebarobot, 2023-10-30:
It seems to be gcc version problem. gcc 13.2.1 is much newer than the gcc 9.4 or gcc 11.2 we use.

wakafa1, 2023-10-30:
It should be fixed in https://github.com/OpenXiangShan/Utility/pull/44

Emin-stack, 2023-10-30:
Thank You! I'll close this issue.

closed 2023-10-30 by Emin-stack