OpenXiangShan/XiangShan #2418: [src/memory/paddr.c:132,paddr_read] isa pmp check failed sh: 1: spike-dasm: not found

Rtr6666, 2023-10-24:
我已经成生成仿真所用的verilog，但是在香山核仿真程序上仿真运行 workload时遇到如下错误：
emu compiled at Oct 24 2023, 10:43:07
Using simulated 32768B flash
isWriteICacheTable0 = 0
isWriteFetchToIBufferTable0 = 0
isWriteIfuWbToFtqTable0 = 0
isWritePrefetchPtrTable0 = 0
isWriteFTQTable0 = 0
isWriteBankConflictTable0 = 0
depth0 = 32
enableDynamicPrefetcher0 = 1
StoreWaitThreshold_0 = 0
nMaxPrefetchEntry0 = 14
isWriteL1MissQMissTable0 = 0
isWriteLoadMissTable0 = 0
isFirstHitWrite0 = 0
isWriteLoadAccessTable0 = 0
isWriteL2TlbPrefetchTable0 = 0
isWriteL1TlbTable0 = 0
isWritePageCacheTable0 = 0
isWritePTWTable0 = 0
isWriteL2TlbMissQueueTable0 = 0
ColdDownThreshold_0 = 12
ForceWriteUpper_0 = 60
ForceWriteLower_0 = 55
StoreBufferThreshold_0 = 7
StoreBufferBase_0 = 4
CorrectMissTrain0 = 0
isWriteInstInfoTable0 = 0
Using simulated 8192MB RAM
The image is /mnt/f/linux/xiangshan/xs-env/XiangShan/ready-to-run/linux.bin
The reference model is /mnt/f/linux/xiangshan/xs-env/NEMU/build/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled. 
[src/memory/paddr.c:132,paddr_read] isa pmp check failed
sh: 1: spike-dasm: not found

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 0010000000 cmtcnt 1 <--
commit group [01]: pc 0000000000 cmtcnt 0
commit group [02]: pc 0000000000 cmtcnt 0
commit group [03]: pc 0000000000 cmtcnt 0
commit group [04]: pc 0000000000 cmtcnt 0
commit group [05]: pc 0000000000 cmtcnt 0
commit group [06]: pc 0000000000 cmtcnt 0
commit group [07]: pc 0000000000 cmtcnt 0
commit group [08]: pc 0000000000 cmtcnt 0
commit group [09]: pc 0000000000 cmtcnt 0
commit group [10]: pc 0000000000 cmtcnt 0
commit group [11]: pc 0000000000 cmtcnt 0
commit group [12]: pc 0000000000 cmtcnt 0
commit group [13]: pc 0000000000 cmtcnt 0
commit group [14]: pc 0000000000 cmtcnt 0
commit group [15]: pc 0000000000 cmtcnt 0

============== Commit Instr Trace ==============
[00] commit pc 0000000010000000 inst 0010029b wen 1 dst 05 data 0000000000000001 idx 000 <--

==============  REF Regs  ==============
  $0: 0x0000000000000000   ra: 0x0000000000000000   sp: 0x0000000000000000   gp: 0x0000000000000000
  tp: 0x0000000000000000   t0: 0x0000000000000001   t1: 0x0000000000000000   t2: 0x0000000000000000
  s0: 0x0000000000000000   s1: 0x0000000000000000   a0: 0x0000000000000000   a1: 0x0000000000000000
  a2: 0x0000000000000000   a3: 0x0000000000000000   a4: 0x0000000000000000   a5: 0x0000000000000000
  a6: 0x0000000000000000   a7: 0x0000000000000000   s2: 0x0000000000000000   s3: 0x0000000000000000
  s4: 0x0000000000000000   s5: 0x0000000000000000   s6: 0x0000000000000000   s7: 0x0000000000000000
  s8: 0x0000000000000000   s9: 0x0000000000000000  s10: 0x0000000000000000  s11: 0x0000000000000000
  t3: 0x0000000000000000   t4: 0x0000000000000000   t5: 0x0000000000000000   t6: 0x0000000000000000
 ft0: 0xe914d7065462f3fc  ft1: 0xb69042b6530acbce  ft2: 0xed0310826c1a9683  ft3: 0x1dba7e1fe29b509b
 ft4: 0xb3647e19fa70386a  ft5: 0x7afe1bc4fb0fef7a  ft6: 0x02090ebca5dc28c5  ft7: 0x086104400dd8273e
 fs0: 0xc3919110875b2aaf  fs1: 0x633fe12a885e5f1e  fa0: 0x0eb4b5ea00529c9e  fa1: 0x0931dd9602ffadae
 fa2: 0x013d9072d28cf875  fa3: 0x82b04d4b3a4b9152  fa4: 0xe72ad7804195e36d  fa5: 0xbb8f6b6064cd9d0f
 fa6: 0x6eeaa984e5bc584d  fa7: 0x38d331c1810632e2  fs2: 0xf4e94301dbc563c1  fs3: 0x54b97ef9a5a6c09f
 fs4: 0x003adb1e2f19a17f  fs5: 0x9e3d15355fd699fa  fs6: 0x8b6b284230ca4700  fs7: 0x40d6674bc97c489f
 fs8: 0xe8329b7c85490dcd  fs9: 0x0b6cf9b13bacd627 fs10: 0x428d8b718b02810e fs11: 0x8cecb3631fc65c0b
 ft8: 0xe61858ee4a385b41  ft9: 0x8d3ba2c674862c2f ft10: 0x9b7335847fe3790e ft11: 0x03c0d4ca985d1b79
pc: 0x0000000000000000 mstatus: 0x0000000a00002000 mcause: 0x0000000000000001 mepc: 0x0000000000000003
                       sstatus: 0x0000000200002000 scause: 0x0000000000000000 sepc: 0x99f584af0eef8c18
satp: 0x0000000000000000
mip: 0x0000000000000000 mie: 0x0000000000000000 mscratch: 0x0000000000000000 sscratch: 0x0000000000000000
mideleg: 0x0000000000000000 medeleg: 0x0000000000000000
mtval: 0x0000000000000003 stval: 0x0000000000000000 mtvec: 0x0000000000000000 stvec: 0x0000000000000000
privilege mode:3  pmp: below
 0: cfg:0x00 addr:0x0000000000000000| 1: cfg:0x00 addr:0x0000000000000000
 2: cfg:0x00 addr:0x0000000000000000| 3: cfg:0x00 addr:0x0000000000000000
 4: cfg:0x00 addr:0x0000000000000000| 5: cfg:0x00 addr:0x0000000000000000
 6: cfg:0x00 addr:0x0000000000000000| 7: cfg:0x00 addr:0x0000000000000000
 8: cfg:0x00 addr:0x0000000000000000| 9: cfg:0x00 addr:0x0000000000000000
10: cfg:0x00 addr:0x0000000000000000|11: cfg:0x00 addr:0x0000000000000000
12: cfg:0x00 addr:0x0000000000000000|13: cfg:0x00 addr:0x0000000000000000
14: cfg:0x00 addr:0x0000000000000000|15: cfg:0x00 addr:0x0000000000000000
priviledgeMode: 3
   mode different at pc = 0x0010000000, right= 0x0000000000000000, wrong = 0x0000000000000003
sstatus different at pc = 0x0010000000, right= 0x0000000000000001, wrong = 0x0000000200002000
   mepc different at pc = 0x0010000000, right= 0x0000000000000003, wrong = 0xb819715d6b902fd0
   sepc different at pc = 0x0010000000, right= 0x0000000200002000, wrong = 0x0000000000000000
mscratch different at pc = 0x0010000000, right= 0x0000000000000003, wrong = 0x0000000000000000
Core 0: ABORT at pc = 0x111216fbc4
instrCnt = 1, cycleCnt = 2096, IPC = 0.000477
Seed=0 Guest cycle spent: 2099 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 1692ms

good-circle, 2023-10-25:
这个问题并不来自于 `sh: 1: spike-dasm: not found`，而是来自于 `[src/memory/paddr.c:132,paddr_read] isa pmp check failed`。

在香山的 difftest 出错后，会检测是否安装了 spike-dasm，如果存在，则通过 spike-dasm 返回指令的反汇编方便 debug。不存在就会报您提到的错误，不会返回反汇编，对执行没有影响。

关于为什么出现 `[src/memory/paddr.c:132,paddr_read] isa pmp check failed` 的报错，目前的信息看上去不足以判断。麻烦提供香山的 commit 版本，编译香山的选项，以便于复现或 debug？

Rtr6666, 2023-10-25:
commit版本 #2414 使用make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32生成香山核，执行./build/emu -i $NOOP_HOME/ready-to-run/linux.bin时产生该错误

closed 2023-10-26 by good-circle