OpenXiangShan/XiangShan #2421: 关于香山代码中的device

changekkk, 2023-10-24:
目前香山代码结构中，有一个文件夹device，是仿真用的外设，这过代码是模拟核外ip的功能吗，这些外设如果需要增加的话，源码如何获取？

wakafa1, 2023-10-25:
这个文件夹中的文件确实是仿真用的模拟外设，包括 SD 卡、Flash、Uart 等；如果想要增加新的模拟外设，欢迎一起参与开发提交 PR

changekkk, 2023-10-25:
请问一下，目前这里已有的模拟外设代码你你们自己开发的一些功能模块吗？是否有开源的代码，另外，这些如果使用现有商业VIP进行替换的话，可以吗？

wakafa1, 2023-10-25:
目前香山流片的 SoC 的外设还是使用商用 IP，因此暂无法开源；如果你有商业 VIP，可以生成香山的 Verilog 代码之后自行连线进行测试

changekkk, 2023-10-28:
目前看着device中的文件代码，实现的是类似 AXI4 **，像是模拟连接外设的一些AXI接口映射，不是模拟外设本身的代码吧？

changekkk, 2023-10-28:
目前仿真里真正模拟外设的代码是不是只有DRAMsim3这个C代码写的一个dram？

wakafa1, 2023-10-28:
> 目前看着device中的文件代码，实现的是类似 AXI4 **，像是模拟连接外设的一些AXI接口映射，不是模拟外设本身的代码吧？

目前香山仿真主要用到的外设为 Flash、SD card、UART、DDR，不仅包含外设的 AXI 口映射，也模拟了外设的行为。但这里所说的“模拟”只是仿真功能模拟，并不是 IP 级的行为模拟

changekkk, 2023-10-30:
那目前如果香山代码编译成可综合代码后（FPGA端用），XSTop.v的顶层代码已经包含了这几个外设功能了吗？我目前看着像是只有几类AXI的接口（包括 DMA、UART、DDR）

wakafa1, 2023-10-30:
> 那目前如果香山代码编译成可综合代码后（FPGA端用），XSTop.v的顶层代码已经包含了这几个外设功能了吗？我目前看着像是只有几类AXI的接口（包括 DMA、UART、DDR）

生成的可综合代码是不会包含这些外设功能的，它们仅供仿真使用

changekkk, 2023-10-30:
明白，谢谢，还有个问题，在DiffTest的src/test/csrc/common下，我看也有一些类似仿真外设的代码，这个和目前device里的这些事配合使用的吗？

wakafa1, 2023-10-30:
> 明白，谢谢，还有个问题，在DiffTest的src/test/csrc/common下，我看也有一些类似仿真外设的代码，这个和目前device里的这些事配合使用的吗？

是的，你可以阅读一下代码查看交互逻辑

changekkk, 2023-10-30:
OK

closed 2023-10-30 by wakafa1