OpenXiangShan/XiangShan #2427: 关于 SMP Load 检查的疑问

GTwhy, 2023-10-25:
Hi，
我发现对于SMP load的值的检查是在 do_instr_commit 中，通过对比goldenmem判断。Load执行过程中，执行refill后，commit前，这期间我理解是允许其他core获得cache的写权限并进行SBuffer->DCache的写入（并在得到resp后更新goldenmem），只是会标记release，但在没有violation的情况下不影响这条load的commit。

是否可能发生如下比对不成功的情况：当前load到的数据是refill中的结果，在从refill到commit之间其他核发生SBufferEvent更新了goldenmem，则load commit进行比对时发现寄存器中的值和goldenmem不符。

例如是否可能存在下图所示情况？假设x是一个内存地址，初始状态为0，store 1,x 代表向x写入1， load x,1 代表load读到内存地址x值为1。
[image]

最后load commit的结果应该可以是1或2？但上述情况下load commit的结果为1，GoldenMem的值被更新为2，NEMU中内存保持初始状态0，所以比对不通过？

也许目前香山在双核情况下没这个问题，但更多核情况下可能存在？不知道我的理解是否正确，请多指教，谢谢！

eastonman, 2023-10-25:
Cache 层级会保证 Core3 30-50 这一段的 ReadShared 与 Core2 的 WriteExclusive 是不会同时发生的。
也即你说的“允许其他core获得cache的写权限并进行SBuffer->DCache的写入”这个是不允许的。
我不是做 Cache 的，如有误请放心指正。

GTwhy, 2023-10-25:
Hi @eastonman，谢谢回复。
确实 Core3 的 ReadShared 与 Core2 的 WriteExclusive 两者不会同时发生。如果 load 在30-40之间完成从 cache line 获取值的操作，则不受 Core3 在40-50之间失去该 cache line 的影响？根据文档和代码来看，Core3 失去 ReadShared 之后，对 LQ 中相关的 load 的 release 置位，用于检测ld-ld violation，但似乎不影响这条load提交？当然也可能影响但我没看到相关细节。

也就是说 load 从cache line 读到值的时间在40前，如果以D$.resp为全局写线性化点，此时的全局序中最后一次 store 就是 Core1 的 store 1, x 。Core 3 此时从全局序中 load 得到1好像也没违反缓存一致性或内存一致性模型（RVWMO ）。

eastonman, 2023-10-25:
你说的有道理，那看来你构造的这个序列应该是合法的。
至于nemu的比对可能要问一下了解的同学。

wakafa1, 2023-10-26:
你说的这个问题确实存在，这也是我们目前默认关闭 SMP Difftest 的原因之一，即 Load 获取数值到 Commit 之间的这段时间内有可能 Golden-Memory 会被其他核改动。我们正在考虑添加状态信息或更改 Load 值的对比时机来 Cover 这种情况，在此之前，多核正确性检查默认退化到 Refill Test，即只检查 DCache Refill 的结果是否与 Golden-Memory 一致。

GTwhy, 2023-10-26:
@wakafa1 感谢回复。
确实退化到 refill check 逻辑上没问题，但不知道还有哪些其他原因导致默认关闭 SMP Difftest？或许可以一起讨论和解决。

wakafa1, 2023-10-26:
你提到的 case 在多核仿真过程中实际发生过，是我们当时关闭 SMP Difftest 的直接原因。此外我们后来 review 这个方案觉得它并不能完全解决 memory consistency 的检查，还需要更多的核内信息来确认不违反 RVWMO，所以正在考虑添加更多探针和计分板来做更详细的检查

GTwhy, 2023-10-30:
好的，我最近也在思考相关问题，后续有进展再交流 ：）

closed 2023-10-30 by GTwhy