OpenXiangShan/XiangShan #2444: vcs仿真不支持Dramsim3

ppx-bit, 2023-10-31:
在vcs.mk中，没有涉及到任何与Dramsim3相关的内容。

请问vcs为什么不支持dramsim3？

是vcs不能仿真带dramsim3的SimTop.v 还是项目组暂未适配？

cebarobot, 2023-10-31:
**English Translation:**
In the vcs.mk file, there is no mention of anything related to Dramsim3.
Why does VCS not support Dramsim3?
Is it that VCS cannot simulate SimTop.v with Dramsim3, or is it that the project team has not yet adapted it?

eastonman, 2023-10-31:
Does VCS support co-simulation with C++?
I know DRAMsim3 is written in C++, so verilator can easily support DRAMsim3.

poemonsense, 2023-12-25:
如果尝试WITH_DRAMSIM3选项打开之后的VCS仿真发现有问题，可以上传相关信息供我们分析

If you have tried VCS with WITH_DRAMSIM3=1, please upload the corresponding logs so that we can provide feedback.

closed 2023-12-25 by poemonsense