OpenXiangShan/XiangShan #2451: make verilog fail (xiangshan[chisel3].runMain subprocess failed make: *** [Makefile:117: build/XSTop.v] Error 1)

Fred-Victor-YANG, 2023-11-02:
**Describe the bug**
make verilog fail
xiangshan[chisel3].runMain subprocess failed
make: *** [Makefile:117: build/XSTop.v] Error 1

**To Reproduce**
git clone --recursive https://github.com/OpenXiangShan/XiangShan.git
make init 
git submodule
git rm -rf (all submodules above)
make init
(no err until now, ping github.com without any trouble, clone all codes correctly)

make verilog

fail

**Expected behavior**
fix and get verilog file 

**Screenshots**
[image]

**Environment (optional, if necessary):**

- 电脑是dell inspiron 15 3511
- 处理器 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz 2.42 GHz
- 机带 RAM 16.0 GB (15.7 GB 可用)
- WSL2 Ubuntu 20.04
- vdisk 208G

**Additional context**
已经看到该 issue 单[Add any other context about the problem here.](https://github.com/OpenXiangShan/XiangShan/issues/837)，但各种重新clone/make init并不能解决问题，问题依然存在。想知道如何才能确定是否是我电脑运存太垃圾导致？

cebarobot, 2023-11-02:
请在运行时监测一下内存占用情况，以排除内存过小的问题。
Please monitor the memory usage during runtime to troubleshoot the issue of low memory usage.

chenguokai, 2023-11-02:
16GB is not sufficient for verilog generation Please add more RAM to your computer or enable swap. To generate verilog, you should have at least 32GB RAM available

Fred-Victor-YANG, 2023-11-03:
有什么办法可以降低内存需求吗？我如果只是想生成一个缩略版的，可以把运存需求控制在10G上下吗？（顺带提一句我在build.sc里找不到MemTotal*的变量...

cebarobot, 2023-11-03:
> 有什么办法可以降低内存需求吗？我如果只是想生成一个缩略版的，可以把运存需求控制在10G上下吗？（顺带提一句我在build.sc里找不到MemTotal*的变量...

两种方法：
* 调大系统的 swap，使得内存 + swap 达到 64 GB 以上。
* （实验性功能）使用 CIRCT 编译，您可以尝试在 `make emu` 时加入 `MFC=1` 参数。

Two methods:
* Increase the system's swap space so that the total memory + swap reaches over 64 GB.
* (Experimental feature) Compile with CIRCT. You can try adding the `MFC=1` parameter during `make emu`.

Fred-Victor-YANG, 2023-11-03:
> 两种方法：
>
> * 调大系统的 swap，使得内存 + swap 达到 64 GB 以上。
> * （实验性功能）使用 CIRCT 编译，您可以尝试在 `make emu` 时加入 `MFC=1` 参数。
>
> Two methods:
>
> * Increase the system's swap space so that the total memory + swap reaches over 64 [...]
我的 wsl disk 已经给到 208G 了，还是一样的错误，是否有可能是我wsl运存受到了电脑本身性能限制呢？

Fred-Victor-YANG, 2023-11-03:
> 两种方法：
>
> * 调大系统的 swap，使得内存 + swap 达到 64 GB 以上。
> * （实验性功能）使用 CIRCT 编译，您可以尝试在 `make emu` 时加入 `MFC=1` 参数。
>
> Two methods:
>
> * Increase the system's swap space so that the total memory + swap reaches over 64 [...]

谢谢！有个另外的问题，我的 wsl disk 已经给到 208G 了，还是一样的错误，是否有可能是我wsl运存受到了电脑本身性能限制呢？

cebarobot, 2023-11-03:
> 谢谢！有个另外的问题，我的 wsl disk 已经给到 208G 了，还是一样的错误，是否有可能是我wsl运存受到了电脑本身性能限制呢？

这不关 wsl disk 大小的事情，您需要调节的是 wsl swap 空间的大小。请阅读微软的[这篇文档](https://learn.microsoft.com/zh-cn/windows/wsl/wsl-config#main-wsl-settings)中关于 SWAP 的设置。

This is not related to the size of the WSL disk. What you need to adjust is the size of the WSL swap space. Please refer to Microsoft's [documentation](https://learn.microsoft.com/en-us/windows/wsl/wsl-config#main-wsl-settings) on SWAP settings.

Fred-Victor-YANG, 2023-11-03:
> 这不关 wsl disk 大小的事情，您需要调节的是 wsl swap 空间的大小。请阅读微软的[这篇文档](https://learn.microsoft.com/zh-cn/windows/wsl/wsl-config#main-wsl-settings)中关于 SWAP 的设置。
>
> This is not related to the size of the WSL disk. What [...]

谢谢！这个方案已经突破上次报错的位置，希望有好结果。作为参考，遇到相同问题的其他开发者可以参考这个流程给 wsl 提供更多的运存：

1. Windows+R 打开 cmd
2. Windows power shell 里先关闭 wsl ：`wsl --shutdown`
3. 继续在 Windows power shell 里打开 user profile：`%UserProfile%` （如果报错了，可以直接在文件夹窗口内打开，通常是 C:\Users\(your_name)\ 目录）
4. 新建文件 .wslconfig，写入以下内容：`[wsl2]
memory=10GB
swap=30GB
localhostForwarding=true`

Fred-Victor-YANG, 2023-11-03:
[image]
完成 swap 拓展wsl空间之后，卡在这里已经三十多分钟了，这情况正常吗？

cebarobot, 2023-11-03:
>  拓展wsl空间之后，卡在这里已经三十多分钟了，这情况正常吗？

这是正常的，chisel3 + defaultConfig 编译时间本身就很长，内存不足使用SWAP将会继续增加编译时间。请耐心等待数个小时。参考 #1453 
It's normal. Chisel3 with defaultConfig will take a long time to compile. Using SWAP other than physical RAM will cost more time. Please wait for a couple of hours. Refer to #1453 

如果您希望加快编译速度：
* 考虑使用 MinimalConfig，即在 `make` 时加入 `CONFIG=MinimalConfig` 参数。但这个配置没有进行严格验证，仅可用于冒烟测试。
* 考虑使用 chisel6，即在 `make` 时加入 `MFC=1`
If you want to accelerate compilation speed:
* Consider to use minimalConfig. Add `CONFIG=MinimalConfig` when `make`.  But this configuration has not been strictly verified and can only be used for smoke testing. 
* Consider to use chisel 6. Add `MFC=1` when `make`.

sequencer, 2023-11-04:
Please switch to circt asap. Since rocket chip got rid of chisel 3p6 now.

Fred-Victor-YANG, 2023-11-06:
> > > 谢谢！有个另外的问题，我的 wsl disk 已经给到 208G 了，还是一样的错误，是否有可能是我wsl运存受到了电脑本身性能限制呢？
> > 
> > 
> > 这不关 wsl disk 大小的事情，您需要调节的是 wsl swap 空间的大小。请阅读微软的[这篇文档](https://learn.microsoft.com/zh-cn/windows/wsl/wsl-config#main-wsl-settings)中关于 SWAP 的设置。
> > This is not related to the size of the WSL disk. What you need to adjust is the size of the WSL swap space. Please refer to Microsoft's [documentation](https://learn.microsoft.com/en-us/windows/wsl/wsl-config#main-wsl-settings) on SWAP settings.
> 
> 谢谢！这个方案已经突破上次报错的位置，希望有好结果。作为参考，遇到相同问题的其他开发者可以参考这个流程给 wsl 提供更多的运存：
> 
> 1. Windows+R 打开 cmd
> 2. Windows power shell 里先关闭 wsl ：`wsl --shutdown`
> 3. 继续在 Windows power shell 里打开 user profile：`%UserProfile%` （如果报错了，可以直接在文件夹窗口内打开，通常是 C:\Users(your_name)\ 目录）
> 4. 新建文件 `.wslconfig`，写入以下内容：`[wsl2] memory=10GB swap=30GB localhostForwarding=true` （当然 mem 和 swap 可以按照本地电脑配置和自己需求修改）（mem是你想给wsl配置的运存，swap顾名思义是mem不够的允许给的交换空间量）
> 5. 重启 wsl 并进入项目文件夹
> 6. make verilog

使用这个方法拓展wsl内存之后，终于成功啦
[image]

closed 2023-11-06 by Fred-Victor-YANG