OpenXiangShan/XiangShan #2465: 设置XSTile中的io_reset_vector之后Tilelink中a通道未取到数据

zhangkanqi, 2023-11-08:
你好！

**问题描述：**
我成功将XSTile中的io_reset_vector信号设置成了0x10000，按照预期，电路复位之后应该要去0x10000这个地址取指令，并更新tilelink a通道的相关信号。但是此时，a通道（包括其他的b、c、d、e通道）的信号都未被更新，如下图所示：
[image]

**疑惑：**
请问为什么a通道的信号没有被更新，a通道没有取到0x10000地址处指令的相关数据呢？要如何解决这个问题呢？

**verilog版本：**
commit 2d7581b5cef8242474fba2305ea8ee11470be115 

**debug波形：**
[XSdebug.zip](https://github.com/OpenXiangShan/XiangShan/files/13291511/XSdebug.zip)

期待收到回复，谢谢:)

wakafa1, 2023-11-08:
地址 0x10000 目前位于香山地址空间中的 Reserved 空间（详见 src/main/scala/xiangshan/backend/fu/PMA.scala），当你试图控制 reset-vector 指向改空间时，会触发访问异常，跳转到 mepc 所在的位置。然而，mepc 在未被修改的情况下默认为 0x0，指向的也是 Reserved 空间，就会导致后续发生无尽的异常，无法获取到第一条指令，与你的波形相符。

Address 0x10000 is currently located in the Reserved space within the XiangShan address space (see src/main/scala/xiangshan/backend/fu/PMA.scala). When attempting to control the reset-vector to point to this space, an access exception is triggered, leading to a jump to the location of mepc. However, mepc defaults to 0x0 when not modified, pointing to the Reserved space as well, causing endless exceptions thereafter, preventing the retrieval of the first instruction, consistent with your waveform.

zhangkanqi, 2023-11-08:
好的，非常感谢！
我还有三点疑惑：
（1）xiangshan默认的reset_vector是多少呢？
（2）依据PMA.scala的设置，只有位于地址[0x80000000, 0x1000000000]的指令才能执行吗？
（3）当访问0x10000地址时，应该会报instruction access fault的异常，请问那根信号上能看到抛出了这个异常呢？我对xiangshan不太熟悉，没找到具体的信号线。

zhangkanqi, 2023-11-08:
> 地址 0x10000 目前位于香山地址空间中的 Reserved 空间（详见 src/main/scala/xiangshan/backend/fu/PMA.scala），当你试图控制 reset-vector 指向该空间时，会触发访问异常，跳转到 mepc 所在的位置。然而，mepc 在未被修改的情况下默认为 0x0，指向的也是 Reserved 空间，就会导致后续发生无尽的异常，无法获取到第一条指令，与你的波形相符。
> 
> Address 0x10000 is currently located in the Reserved space within the XiangShan address space (see src/main/scala/xiangshan/backend/fu/PMA.scala). When attempting to control the reset-vector to point to this space, an access exception is triggered, leading to a jump to the location of mepc. However, mepc defaults to 0x0 when not modified, pointing to the Reserved space as well, causing endless exceptions thereafter, preventing the retrieval of the first instruction, consistent with your waveform.

我尝试将reset_vector设成了0x8000 0000，但a通道的相关信号还是没有被更新。

huxuan0307, 2023-11-09:
> 好的，非常感谢！ 我还有三点疑惑： （1）xiangshan默认的reset_vector是多少呢？ （2）依据PMA.scala的设置，只有位于地址[0x80000000, 0x1000000000]的指令才能执行吗？ （3）当访问0x10000地址时，应该会报instruction access fault的异常，请问那根信号上能看到抛出了这个异常呢？我对xiangshan不太熟悉，没找到具体的信号线。
> 
> I have three more questions: (1) what is the default reset_vector of XiangShan? (2) According to settings in PMA.scala, is address range [0x80000000, 0x1000000000] the only executable range? (3) When accessing 0x10000, the processor should report instruction access fault, which signal may I check if this exception is triggered?

(1)顶层模块是XSTile，看代码可以指导reset_vector从IO传入，是可以由外部连线指定的。[SimTop.scala:58](https://github.com/OpenXiangShan/XiangShan/blob/650ccc40631f1ef966336a03bcce8c815367a808/src/test/scala/top/SimTop.scala#L58)，这里有testbench对reset_vector的仿真时默认值0x10000000。
(2)PMA的设置里有多个可执行地址段，例如[0x10000000,0x20000000),[0x38020000~0x38021000)等。
(3)异常可以在decode阶段开始的exceptionVec里看到。异常向量编号可以查[riscv privileged spec](https://github.com/riscv/riscv-isa-manual/releases/tag/Priv-v1.12)

(1)The top module is XSTile, looking at the code, I found that reset_vector is passed in from IO, which can be specified by the external connection. [SimTop.scala:58](https://github.com/OpenXiangShan/XiangShan/blob/650ccc40631f1ef966336a03bcce8c815367a808/src/test/scala/top/SimTop.scala#L58), here there is the default value of 0x10000000 when testbench simulates on reset_vector.
(2)There are several executable address segments in PMA's settings, such as [0x10000000,0x20000000),[0x38020000~0x38021000), etc.
(3) Exceptions can be seen in the exceptionVec at the beginning of the decode phase. You can check exception vector number in  [riscv privileged spec](https://github.com/riscv/riscv-isa-manual/releases/tag/Priv-v1.12)

zhangkanqi, 2023-11-28:
Thanks for your reply.
mmio-related signals are controlled by XSTop and SimTop, So I must run emulational verilog(which has SimTop) not synthetical verilog.

closed 2023-11-28 by zhangkanqi