OpenXiangShan/XiangShan #2467: make verilog   fudian.compile compilation failed

yuanerandlee2, 2023-11-09:
执行如下命令，出现fudian.compile compilation failed
cd XiangShan
export NOOP_HOME=$(pwd)
git checkout nanhu-clockdiv2
make init
make clean
make verilog NUM_CORES=4

系统版本: 
Linux hjcs02 5.15.0-88-generic #98~20.04.1-Ubuntu SMP Mon Oct 9 16:43:45 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
java version:
openjdk version "11.0.20.1" 2023-08-24

具体信息如下：
root@hjcs02:/home/soft01/xiangshan/xs-env_4core/XiangShan# make init
git submodule update --init
cd rocket-chip && git submodule update --init api-config-chipsalliance hardfloat
root@hjcs02:/home/soft01/xiangshan/xs-env_4core/XiangShan# make clean
make -C ./difftest clean
make[1]: Entering directory '/home/soft01/xiangshan/xs-env_4core/XiangShan/difftest'
rm -rf /home/soft01/xiangshan/xs-env_4core/XiangShan/difftest/../sim /home/soft01/xiangshan/xs-env_4core/XiangShan/build/simv-compile
rm -rf /home/soft01/xiangshan/xs-env_4core/XiangShan/difftest/../build
make[1]: Leaving directory '/home/soft01/xiangshan/xs-env_4core/XiangShan/difftest'
rm -rf ./build
root@hjcs02:/home/soft01/xiangshan/xs-env_4core/XiangShan# make verilog NUM_CORES=4
mkdir -p build
mill -i XiangShan.runMain top.TopMain -td build                      \
	--config DefaultConfig --full-stacktrace --output-file XSTop.v    \
	--infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf \
	--gen-mem-verilog full --num-cores 4             \
	--disable-all --remove-assert --fpga-platform
[177/214] fudian.compile 
[info] compiling 17 Scala sources to /home/soft01/xiangshan/xs-env_4core/XiangShan/out/fudian/compile/dest/classes ...
[error] /home/soft01/xiangshan/xs-env_4core/XiangShan/fudian/src/main/scala/fudian/Generator.scala:13:29: value getPackageName is not a member of Class[T]
[error]     val pkg = this.getClass.getPackageName
[error]                             ^
[error] one error found
1 targets failed
fudian.compile Compilation failed
make: *** [Makefile:73: build/XSTop.v] Error 1
root@hjcs02:/home/soft01/xiangshan/xs-env_4core/XiangShan# java -version
openjdk version "11.0.20.1" 2023-08-24
OpenJDK Runtime Environment (build 11.0.20.1+1-post-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 11.0.20.1+1-post-Ubuntu-0ubuntu120.04, mixed mode, sharing)
root@hjcs02:/home/soft01/xiangshan/xs-env_4core/XiangShan# free -h
              total        used        free      shared  buff/cache   available
Mem:          251Gi       7.6Gi       211Gi        54Mi        32Gi       242Gi
Swap:         2.0Gi          0B       2.0Gi
root@hjcs02:/home/soft01/xiangshan/xs-env_4core/XiangShan#

cebarobot, 2023-11-09:
看起来子模块没有检出到这个分支的指定版本，请在运行下面的命令后再尝试编译：
It seems that submodule was not checkouted to the specified version of this branch. Please try compiling after run the following commands:
```
git submodule update --init --recursive
```

yuanerandlee2, 2023-11-09:
还是同样的错误

poemonsense, 2023-11-09:
请参考https://github.com/OpenXiangShan/XiangShan/issues/837#issuecomment-867667342 里面的几个步骤

比如，是否在java 8的环境中尝试跑mill，之后才切换回java 11？

yuanerandlee2, 2023-11-09:
i have tested it with all the solution mentioned at #837 , also the same error

cebarobot, 2023-11-09:
我这边没能浮现您的问题，能否运行一下 `mill --version` 并提供运行结果？

yuanerandlee2, 2023-11-09:
是否在java 8的环境中尝试跑mill，之后才切换回java 11  how to test this?

yuanerandlee2, 2023-11-09:
> 我这边没能浮现您的问题，能否运行一下 `mill --version` 并提供运行结果？

root@hjcs02:/home/soft01/xs_ljh/xs-env/XiangShan# mill --version
Mill Build Tool version 0.9.8
Java version: 1.8.0_101, vendor: Oracle Corporation, runtime: /usr/local/java/jre
Default locale: en_US, platform encoding: UTF-8
OS name: "Linux", version: 5.15.0-88-generic, arch: amd64

cebarobot, 2023-11-09:
> root@hjcs02:/home/soft01/xs_ljh/xs-env/XiangShan# mill --version Mill Build Tool version 0.9.8 Java version: 1.8.0_101, vendor: Oracle Corporation, runtime: /usr/local/java/jre Default locale: [...]

看起来 mill 运行时调用的 Java 版本是 Java8 呢……运行 `echo $JAVA_HOME` 命令看看这个环境变量指向哪里？然后试试把他改成 OpenJDK11 的目录，然后再运行 `mill --version` 看看。

yuanerandlee2, 2023-11-09:
change the $JAVA_HOME to the directory of java11 can work ,thank you.

closed 2023-11-09 by yuanerandlee2