OpenXiangShan/XiangShan #2472: 请教一个关于前端分支预测器的问题

HarsonLau, 2023-11-11:
在前端代码实现中，我注意到 [FullBranchPrediction](https://github.com/OpenXiangShan/XiangShan/blob/650ccc40631f1ef966336a03bcce8c815367a808/src/main/scala/xiangshan/frontend/FrontendBundle.scala#L426) 已经足以描述一个取指块，但是在[BranchPredictionBundle](https://github.com/OpenXiangShan/XiangShan/blob/650ccc40631f1ef966336a03bcce8c815367a808/src/main/scala/xiangshan/frontend/FrontendBundle.scala#L572C7-L572C29) 中，PC，有效位，FullBranchPrediction等信息被 复制了4次。我想请教一下为什么需要复制4次，这样设计背后的原因是什么？

chenguokai, 2023-11-11:
这一设计是为了优化时序，复制信号能够限制每个信号的扇出
This design is for timing optimization. By duplicating signals, fan out is decreased for every single signal.

Tang-Haojin, 2023-12-13:
我们应该已经解决了您的问题。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

Your issue may have been solved. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2023-12-13 by Tang-Haojin