OpenXiangShan/XiangShan #2475: VCS仿真报错Can not open 'ram.bin'

zhangkanqi, 2023-11-13:
**问题描述：**
按照https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/vcs/ 这里的教程执行
`./difftest/simv [+workload=WorkloadName.bin] [+flash=Flash.bin] [+dump-wave] [+diff=Nemu-so] [+no-diff] [+max-cycles=MaxCycleNum]` 报错。具体报错信息如下：
>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv [+workload=WorkloadName.bin] [+flash=Flash.bin] [+dump-wave] [+diff=Nemu-so] [+no-diff] [+max-cycles=MaxCycleNum]
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 13 14:54 2023
simv compiled at Nov 13 2023, 12:52:23
The image is ram.bin
Using simulated 8192MB RAM
Can not open 'ram.bin'
simv: /home/xxxx-DomainA/xs-env/XiangShan/difftest/src/test/csrc/common/ram.cpp:161: void init_ram(const char*): Assertion `0' failed.
An unexpected termination has occurred in ./difftest/simv due to a signal: Aborted
Hostname h00b700
Command line: ./difftest/simv [+workload=WorkloadName.bin] [+flash=Flash.bin] [+dump-wave] [+diff=Nemu-so] [+no-diff] [+max-cycles=MaxCycleNum]
Child process exited with status 127: /home/limiao-DomainA/VCS-MX2018.09-SP2/linux64/bin/cbug-gdb-64/bin/gdb -n --batch -x /tmp/.stack_annotator_gdb_cmd_file_Na6SLf ./difftest/simv 3328265 < /dev/null &> /tmp/.stack_annotator_gdb_output_file_TSTyqO
--- Stack trace follows:
Dumping VCS Annotated Stack:
Process VmPeak: 9375872 kb, VmSize: 9375836 kb
System Free Memory: 429223036 kb, System Free Swap: 0 kb
No context available

**环境：**
XiangShan：cb6e5d3cb
操作系统：Ubuntu 23.04
difftest：6107002b
vcs：O-2018.09-SP2_Full64（已解决`make simv RELEASE=1`时的报错https://blog.csdn.net/qianniuwei321/article/details/127303086）

cebarobot, 2023-11-13:
emmmmmm……命令行说明中的方括号 `[]` 是可选参数的意思，在执行命令时不需要加方括号 `[]`。
errrrrr...The square brackets [] in the command line instructions indicate optional parameters, and they should not be included when executing the command.

如果您只是进行 vcs 的尝试，您大概率不需要加上参数。类似这样：
If you are just trying VCS, you probably don't need to include additional parameters. Like this:
```
./difftest/simv
```
如果您希望手动指定内存镜像（记得将`path_to/workload.bin`替换成实际路径）：
If you want to manually specify the memory image, you can do it like this (Remember to replace path_to/workload.bin with the actual file path):
```
./difftest/simv +workload=path_to/workload.bin
```

zhangkanqi, 2023-11-13:
Oops！(*/ω＼*)

那请问执行波形的参数是什么呢？
>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv +workload=ready-to-run/microbench.bin +flash=Flash.bin +dump-wave +diff=Nemu-so
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 13 16:06 2023
unknown wave file format, want [vpd, fsdb] but:
$finish called from file "/home/xxxx-DomainA/xs-env/XiangShan/difftest/src/test/vsrc/vcs/top.v", line 63.
$finish at simulation time                    0
           V C S   S i m u l a t i o n   R e p o r t
Time: 0 ns
CPU Time:      1.030 seconds;       Data structure size: 160.8Mb
Mon Nov 13 16:06:02 2023

cebarobot, 2023-11-13:
> Oops！(*/ω＼*)
> 
> 那请问指定波形的参数是什么呢？
> 

请阅读[这篇文档](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/vcs/)的中“simv 参数说明如下”的第三部分：
Please RTFM (Read [The Friendly Manual](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/vcs/)):
> +dump-wave: By default, simv will not dump waveform. Waveform file is simv.vpd.

zhangkanqi, 2023-11-13:
I should specify that `+dump-wave=vpd`.  But where is Flash.bin if there is?
>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv +workload=ready-to-run/microbench.bin +flash=Flash.bin +dump-wave=vpd +diff=Nemu-so
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 13 16:20 2023
VCD+ Writer O-2018.09-SP2_Full64 Copyright (c) 1991-2018 by Synopsys Inc.
ram image:ready-to-run/microbench.bin
flash image:Flash.bin
diff-test ref so:Nemu-so
simv compiled at Nov 13 2023, 12:52:23
The image is ready-to-run/microbench.bin
Using simulated 8192MB RAM
Using simulated 32768B flash
[info]use Flash.bin as flash bin
**[error] flash img not found**

zhangkanqi, 2023-11-13:
Moreover, can u show one available  value of  `+diff`  parameter as an example？I tried `Nemu-so` and `build-verilator/emu`, but it didn't work.

>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv +workload=ready-to-run/microbench.bin  +dump-wave=vpd **+diff=Nemu-so**
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 13 16:27 2023
VCD+ Writer O-2018.09-SP2_Full64 Copyright (c) 1991-2018 by Synopsys Inc.
ram image:ready-to-run/microbench.bin
diff-test ref so:Nemu-so
simv compiled at Nov 13 2023, 12:52:23
The image is ready-to-run/microbench.bin
Using simulated 8192MB RAM
Using simulated 32768B flash
[warning]no valid flash bin path, use preset flash instead
NemuProxy using Nemu-so
Nemu-so: cannot open shared object file: No such file or directory
simv: /home/xxxx-DomainA/xs-env/XiangShan/difftest/src/test/csrc/difftest/refproxy.cpp:55: NemuProxy::NemuProxy(int, size_t): Assertion `0' failed.

>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv +workload=ready-to-run/microbench.bin  +dump-wave=vpd **+diff=build-verilator/emu**
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 13 16:29 2023
VCD+ Writer O-2018.09-SP2_Full64 Copyright (c) 1991-2018 by Synopsys Inc.
ram image:ready-to-run/microbench.bin
diff-test ref so:build-verilator/emu
simv compiled at Nov 13 2023, 12:52:23
The image is ready-to-run/microbench.bin
Using simulated 8192MB RAM
Using simulated 32768B flash
[warning]no valid flash bin path, use preset flash instead
NemuProxy using build-verilator/emu
build-verilator/emu: cannot dynamically load position-independent executable
simv: /home/xxxx-DomainA/xs-env/XiangShan/difftest/src/test/csrc/difftest/refproxy.cpp:55: NemuProxy::NemuProxy(int, size_t): Assertion `0' failed.

cebarobot, 2023-11-13:
> I should specify that `+dump-wave=vpd`. But where is Flash.bin if there is?

If you are not sure to use which flash rom, please consider not add `+flash=`. As document says, 
如果您不确定您要用什么固件（Flash.bin），您可以考虑不加 `+flash=` 选项。正如文档所说：
> By default, simv uses built-in flash, which will jump to 0x20_0000_0000.
> 默认情况下，simv 将使用内置的 flash，这个 flash 会跳转到 0x20_0000_0000 以执行后续的程序。

> Moreover, can u show one available value of +diff parameter as an example？I tried Nemu-so and build-verilator/emu, but it didn't work.

The diff here is [**NEMU**](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/nemu/), rather than emu generated by Verilator. You can compile your onw NEMU according to the document, or just use `ready-to-run/riscv64-nemu-interpreter-so`.
这里的 diff 需要使用的是 [**NEMU**](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/nemu/)，而不是 Verilator 生成的模拟器 emu。您可以根据文档自己编译 NEMU，或是使用 `ready-to-run/riscv64-nemu-interpreter-so`。

zhangkanqi, 2023-11-13:
Why there are many errors, does that meet expectations？

>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv +workload=ready-to-run/microbench.bin  +dump-wave=vpd +diff=ready-to-run/riscv64-nemu-interpreter-so
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 13 16:51 2023
VCD+ Writer O-2018.09-SP2_Full64 Copyright (c) 1991-2018 by Synopsys Inc.
ram image:ready-to-run/microbench.bin
diff-test ref so:ready-to-run/riscv64-nemu-interpreter-so
simv compiled at Nov 13 2023, 12:52:23
The image is ready-to-run/microbench.bin
Using simulated 8192MB RAM
Using simulated 32768B flash
[warning]no valid flash bin path, use preset flash instead
NemuProxy using ready-to-run/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled.
[NEMU] Can not find flash image: (null)
[NEMU] Use built-in image instead
[src/device/io/mmio.c:38,add_mmio_map] Add mmio map 'flash' at [0x0000000010000000, 0x00000000100fffff]
ERROR: ram wIdx = 0x4db80cf8 out of bound!
ERROR: ram wIdx = 0x4db80cf9 out of bound!
ERROR: ram wIdx = 0x4db80cfa out of bound!
ERROR: ram wIdx = 0x4db80cfb out of bound!
ERROR: ram wIdx = 0x4db80cfc out of bound!
ERROR: ram wIdx = 0x4db80cfd out of bound!
ERROR: ram wIdx = 0x4db80cfe out of bound!
ERROR: ram wIdx = 0x4db80cff out of bound!
ERROR: ram wIdx = 0x48e40f68 out of bound!
ERROR: ram wIdx = 0x48e40f69 out of bound!
ERROR: ram wIdx = 0x48e40f6a out of bound!
ERROR: ram wIdx = 0x48e40f6b out of bound!
ERROR: ram wIdx = 0x48e40f6c out of bound!
ERROR: ram wIdx = 0x48e40f6d out of bound!
ERROR: ram wIdx = 0x48e40f6e out of bound!
ERROR: ram wIdx = 0x48e40f6f out of bound!
ERROR: ram wIdx = 0x531c1088 out of bound!
ERROR: ram wIdx = 0x531c1089 out of bound!
ERROR: ram wIdx = 0x531c108a out of bound!
ERROR: ram wIdx = 0x531c108b out of bound!
ERROR: ram wIdx = 0x531c108c out of bound!
ERROR: ram wIdx = 0x531c108d out of bound!
ERROR: ram wIdx = 0x531c108e out of bound!
ERROR: ram wIdx = 0x531c108f out of bound!
ERROR: ram wIdx = 0x5cd80ff0 out of bound!
ERROR: ram wIdx = 0x5cd80ff1 out of bound!
ERROR: ram wIdx = 0x5cd80ff2 out of bound!
ERROR: ram wIdx = 0x5cd80ff3 out of bound!
ERROR: ram wIdx = 0x5cd80ff4 out of bound!
ERROR: ram wIdx = 0x5cd80ff5 out of bound!
ERROR: ram wIdx = 0x5cd80ff6 out of bound!
ERROR: ram wIdx = 0x5cd80ff7 out of bound!
======= Running MicroBench ERROR: ram wIdx = 0x433c0ef0 out of bound!
ERROR: ram wIdx = 0x433c0ef1 out of bound!
ERROR: ram wIdx = 0x433c0ef2 out of bound!
ERROR: ram wIdx = 0x433c0ef3 out of bound!
ERROR: ram wIdx = 0x433c0ef4 out of bound!
ERROR: ram wIdx = 0x433c0ef5 out of bound!
ERROR: ram wIdx = 0x433c0ef6 out of bound!
ERROR: ram wIdx = 0x433c0ef7 out of bound!
[input *test*] ERROR: ram wIdx = 0x72f20d30 out of bound!
ERROR: ram wIdx = 0x72f20d31 out of bound!
ERROR: ram wIdx = 0x72f20d32 out of bound!
ERROR: ram wIdx = 0x72f20d33 out of bound!
ERROR: ram wIdx = 0x72f20d34 out of bound!
ERROR: ram wIdx = 0x72f20d35 out of bound!
ERROR: ram wIdx = 0x72f20d36 out of bound!
ERROR: ram wIdx = 0x72f20d37 out of bound!
=======
[WARNING] difftest store queue overflow

wakafa1, 2023-11-17:
经过测试，我们发现当前 VCS 仿真不能很好地识别代码中的 clock gating 原语，我们会在之后进行修复。在此之前，你可以修改代码关闭 L3 的 clock gating 以实现正确的仿真，代码修改如下：

https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/top/Configs.scala#L286  将 true 改为 false

zhangkanqi, 2023-11-17:
修改完重新编译之后，执行`/difftest/simv +workload=ready-to-run/microbench.bin +dump-wave=vpd +diff=ready-to-run/riscv64-nemu-interpreter-so`应该是成功了，但就是跑得非常慢。

>xxxx@h00b700:~/xs-env/XiangShan$ ./difftest/simv +workload=ready-to-run/microbench.bin +dump-wave=vpd +diff=ready-to-run/riscv64-nemu-interpreter-so
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Nov 17 16:03 2023
VCD+ Writer O-2018.09-SP2_Full64 Copyright (c) 1991-2018 by Synopsys Inc.
ram image:ready-to-run/microbench.bin
diff-test ref so:ready-to-run/riscv64-nemu-interpreter-so
simv compiled at Nov 13 2023, 12:52:23
The image is ready-to-run/microbench.bin
Using simulated 8192MB RAM
Using simulated 32768B flash
[warning]no valid flash bin path, use preset flash instead
NemuProxy using ready-to-run/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled.
[NEMU] Can not find flash image: (null)
[NEMU] Use built-in image instead
[src/device/io/mmio.c:38,add_mmio_map] Add mmio map 'flash' at [0x0000000010000000, 0x00000000100fffff]
======= Running MicroBench [input *test*] =======
[WARNING] difftest store queue overflow
[qsort] Quick sort: * Passed.
[queen] Queen placement: * Passed.

wakafa1, 2023-11-17:
所以我们更多时候使用 verilator 仿真 : )
So we use verilator simulation more often : )

zhangkanqi, 2023-11-17:
我用VCS仿真主要是想在dve里对着波形追verilog代码来更好的解决 #2487 ，但是目前还没什么头绪QAQ

Tang-Haojin, 2023-12-13:
我们看到您的问题似乎已经在 #2487 中得到了解决，因此我们将关闭本 issue。如有需要，欢迎再次打开 issue 或提交新的 issue。

We see that the root issue seems to have been resolved in #2487, so we close this issue. If necessary, please re-open the issue or file a new issue.

closed 2023-12-13 by Tang-Haojin