OpenXiangShan/XiangShan #2481: 关于TL-Test使用

changekkk, 2023-11-15:
你好，请教一下，关于TL-Test在不同验证层级的使用，目前提供的工程里看到，有模块级和子系统级的使用，SOC级的应用，也就是说只保留monitor的bus rule checker功能，master/slave  agent关闭的使用方式，目前有实际应用吗？或者说，关闭master和slave  agent的方式是什么？

Ivyfeather, 2023-11-15:
目前我们的 TL-Test 只能支持模块级/子系统 (L2-L3) 级，由 TL-Test 主动发出请求。暂时不支持 SOC 级只使用 monitor 和 checker 的方案，因为框架里的对计分板 ScoreBoard 的更新和 Agent 发请求逻辑是紧耦合的。如果想要支持，可能需要修改一下这部分代码，添加根据外部请求更新 ScoreBoard 的逻辑，并添加将总线请求传入 TL-Test 的硬件逻辑。

如果想验证香山 SOC 级的一致性，目前我们的 difftest 框架中有关于一致性的检查，它维护了一个 Golden Memory，在原子指令/Store Buffer写入L1时更新（update_goldenmem），并在向 L1 重填时进行数据检查（refill test），可以查看一下这部分代码。

Ivyfeather, 2023-11-15:
Currently, our TL-Test only supports unit-level/subsystem (L2-L3) testing. We temporarily do not support a solution at the SOC level that only utilizes monitors and checkers, because the logic for updating the scoring board and sending requests from the agent is tightly coupled within the framework. To support such a solution, it may be necessary to modify the code by adding logic to update the scoring board based on external requests and incorporate the hardware logic of passing bus requests to TL-Test.

If you wish to verify the consistency of the Xiangshan SOC level, our difftest framework includes consistency checks. It maintains a Golden Memory that is updated when atomic instructions/Store Buffers write to L1 (update_goldenmem), and performs data checks during L1 refill (refill test). You can refer to the corresponding code for further details.

changekkk, 2023-11-15:
好的，同时我看到咱们目前在昆明湖架构新引入的 CHI-Test，这个是可以支持SOC层级的验证吗？

Ivyfeather, 2023-11-16:
目前的 CHI-Test 也还不支持 SOC 层次的验证

changekkk, 2023-11-16:
目前CHI-Test代码可以看到吗？

Ivyfeather, 2023-11-16:
目前 CHI-Test 还尚未完成，所以暂时看不到