OpenXiangShan/XiangShan #2487: No data response at address 0x10000000 after set reset_vector

zhangkanqi, 2023-11-17:
**问题描述：**
使用香山的综合verilog代码（删除了XSTop，使top module是XSTile），reset_vector设置为0x1000 0000。
在mmio中，0x1000 0000处的地址请求成功发出，但mmio_grant_valid信号并没有拉高，也没有返回data，请问是怎么回事呢？mmio_grant_valid信号能拉高的根本条件是什么呢？

0x1000 0000处设置的数据如下:
[image]

波形如下，可以看到mmio_grant_valid并没有拉高：
[image]

**verilog源码和波形：**
[XiangShan-syn-fail.zip](https://github.com/OpenXiangShan/XiangShan/files/13387040/XiangShan-syn-fail.zip)

poemonsense, 2023-11-17:
mmio_grant_ready有效，代表CPU可以接收外面返回的mmio数据；mmio_grant_valid无效，代表CPU外部总线没有正确返回请求数据，需要检查您的TB环境是否有正确响应CPU发出的读写请求

zhangkanqi, 2023-11-28:
Thanks for your reply.
mmio-related signals are controlled by XSTop and SimTop, So I must run  emulational verilog(which has SimTop) not synthetical verilog.

closed 2023-11-28 by zhangkanqi