OpenXiangShan/XiangShan #2492: "requirement failed" when building emu with MinimalSimConfig

camel-cdr, 2023-11-20:
I tried to build ` make emu CONFIG=MinimalSimConfig`, but got the following error:

```
Exception in thread "main" java.lang.IllegalArgumentException: requirement failed: Diplomacy has detected a problem with your graph:
The following node appears left of a := 0 times and right of a := 1 times.
Either the number of bindings on both sides of the node match, or connect this node by left-hand side of :*= or right-hand side of :=*
adapter logger.node node:
parents: logger/misc/core_with_l2/l_soc
locator:  (TLLogger.scala:183:30)

0 inward nodes bound: []
1 outward nodes bound: [once-busPMU.node]

      at ... ()
      at freechips.rocketchip.diplomacy.MixedAdapterNode.resolveStar(Nodes.scala:1453)
      at freechips.rocketchip.diplomacy.MixedNode.liftedTree1$1(Nodes.scala:1057)
       at freechips.rocketchip.diplomacy.MixedNode.x$27$lzycompute(Nodes.scala:1028)
       at freechips.rocketchip.diplomacy.MixedNode.x$27(Nodes.scala:1027)
       at freechips.rocketchip.diplomacy.MixedNode.oPortMapping$lzycompute(Nodes.scala:1027)
       at freechips.rocketchip.diplomacy.MixedNode.oPortMapping(Nodes.scala:1027)
       at freechips.rocketchip.diplomacy.MixedNode.$anonfun$iDirectPorts$1(Nodes.scala:1105)
       at scala.collection.immutable.List.flatMap(List.scala:366)
       at freechips.rocketchip.diplomacy.MixedNode.iDirectPorts$lzycompute(Nodes.scala:1103)
       at freechips.rocketchip.diplomacy.MixedNode.iDirectPorts(Nodes.scala:1103)
       at freechips.rocketchip.diplomacy.MixedNode.iPorts$lzycompute(Nodes.scala:1143)
       at freechips.rocketchip.diplomacy.MixedNode.iPorts(Nodes.scala:1143)
...
1 targets failed
XiangShan.test.runMain subprocess failed

```

My machine only has 16G of RAM, so I made sure to allocate enough swap space  (an extra 64G).

I used the following `Dockerfile` to do the build:

```Dockerfile
FROM ubuntu:22.04

RUN apt-get update \
	&& apt-get install -y build-essential clang libclang-dev llvm-dev cmake \
	libspdlog-dev vim git libmlpack-dev curl wget time default-jre default-jdk \
	&& apt-get clean \
	&& rm -rf /var/lib/apt/lists/*

RUN git clone --recursive https://github.com/OpenXiangShan/xs-env
RUN cd xs-env && ./update-submodule.sh
RUN cd xs-env && sed 's/source .*$//g' -i ./setup-tools.sh && ./setup-tools.sh
RUN cd xs-env && sed 's/sudo //g' -i ./install-verilator.sh && . ./install-verilator.sh
RUN cd xs-env && . ./setup.sh

WORKDIR /xs-env
RUN . ./env.sh && make -C XiangShan init
RUN . ./env.sh && make -C XiangShan emu CONFIG=MinimalSimConfig -j 5
```

wakafa1, 2023-11-21:
We haven't maintained MinimalSimConfig for a long time and we will check it out later. Before that, please try CONFIG=MinimalConfig

camel-cdr, 2023-11-21:
I tried that, but I stop it after it didn't give any new output for two hours. I'll try running it again over night.

I also tried building `CONFIG=MinimalConfig MFC=1`, because I read `MFC=1` might be faster, but that errored:

```
 Error: firtool failed.
 ExitCode:
 1
 STDOUT:

 STDERR:
 firtool: Unknown command line argument '-repl-seq-mem-circuit=top.SimTop'.  Try: 'firtool --help'
 firtool: Did you mean '--repl-seq-mem-file=top.SimTop'?
```

wakafa1, 2023-11-22:
Please make sure your XiangShan repo is up-to-date, or have #2372 included at least

camel-cdr, 2023-11-22:
I used the latest branches for MFC, that should've been done by the `./update-submodule.sh`, but I the build finished with `CONFIG=MinimalConfig` and without MFC.
:+1:

cebarobot, 2023-11-22:
> I used the latest branches for MFC, that should've been done by the `./update-submodule.sh`, but I the build finished with `CONFIG=MinimalConfig` and without MFC. 👍

Could you please provide the commit id of your XiangShan repo?

camel-cdr, 2023-11-22:
Looks like you where right, the XiangShan repo was on a commit from September.

closed 2023-11-22 by camel-cdr