OpenXiangShan/XiangShan #2498: 请教与综合相关的问题

menglinhan, 2023-11-22:
您好，我这边已经通过make verilog和make simv生成了两版代码，现在是自己搭建环境跑了vcs仿真（使用的是make simv生成的代码，用XSTop级集成），我有两个问题：
1） 我使用make simv生成的代码自XSTop级综合是否可以？
2） 我替换memory的话，是不是所有memory均在array_*_ext.v文件中？

wakafa1, 2023-11-23:
问题 1：可以，XSTop 是设计顶层
问题 2：是的

Question 1: Yes, XSTop is the top level of the design
Question 2: Yes

menglinhan, 2023-11-23:
谢谢你的回复，从香山的文档我找到了下面的信息“NANHU targets 2GHz@14nm, and 2.4GHz~2.8GHz@7nm.<br>南湖的目标是在14nm工艺下频率达到2GHz，在7nm工艺下频率达到2.4GHz到2.8GHz。”，请问14nm下的2GHz是在什么电压下的，该电压是最高电压吗？

thanks for your reply! I found follow information from Xiangshan's Doc "NANHU targets 2GHz@14nm, and 2.4GHz~2.8GHz@7nm.<br>南湖的目标是在14nm工艺下频率达到2GHz，在7nm工艺下频率达到2.4GHz到2.8GHz。". What voltage is 2GHz under 14nm, and is it the highest voltage?

wakafa1, 2023-11-23:
该评估基于 TT corner 0.9V

This evaluation is based on TT corner 0.9V

menglinhan, 2023-11-23:
好的，多谢！

OK，Thanks！

menglinhan, 2023-11-23:
您好，再请教一下，温度是多少呢？25 度吗？

What is the temperature? Is it 25 degrees?

wakafa1, 2023-11-23:
85度，顺便一提温度对频率影响不大，85跟25基本一样的，主要影响功耗

85 degrees. By the way, temperature has little effect on frequency. 85 is basically the same as 25, and it mainly affects power consumption.

menglinhan, 2023-11-23:
好的，多谢！

OK，Thanks！

closed 2023-11-24 by wakafa1

menglinhan, 2023-11-30:
您好，关于综合我再咨询几个问题，除了memory，有其他标准单元需要替换后才能综合吗？我有看到clkgate和一些sync模块。

wubo0822, 2025-05-10:
> 您好，我这边已经通过make verilog和make simv生成了两版代码，现在是自己搭建环境跑了vcs仿真（使用的是make simv生成的代码，用XSTop级集成），我有两个问题： 1） 我使用make simv生成的代码自XSTop级综合是否可以？ 2） 我替换memory的话，是不是所有memory均在array_*_ext.v文件中？
> 
> Hello, I've generated two versions of code using `make verilog` and `make simv`. I've set up my own environment to run VCS simulation (using the code generated by `make simv` and utilizing XSTop for top-level integration). I have two questions:
> 
> 1. Can I perform synthesis from the code generated by 'make simv' for XSTop-level integration?
> 2. If I want to replace the memory, should I modify all memory instances in the `array_*_ext.v` file?

楼主前辈您好，我也在尝试基于mark verilog的代码进行dc综合，目前卡在了array_20_ext.v这个文件，请问您是如何替换memory的呢？其他标准单元需要替换吗？

menglinhan, 2025-07-10:
> 楼主前辈您好，我也在尝试基于mark verilog的代码进行dc综合，目前卡在了array_20_ext.v这个文件，请问您是如何替换memory的呢？其他标准单元需要替换吗？

可以用smic或者tsmc的生成的memory替换，你们应该有mem生成工具吧？