OpenXiangShan/XiangShan #2506: 请问香山里面自带的XSDebug的信息怎么打印出来？生成的log文件里面好像只有性能计数器的信息

YudeFang, 2023-11-24:
[image]

Tang-Haojin, 2023-11-24:
可以试试给 Makefile 里的 `DEBUG_ARGS` 加上 `--enable-log` 参数。

You may try to append `--enable-log` to `DEBUG_ARGS` in the Makefile.

``` diff
diff --git a/Makefile b/Makefile
index 205ba9450..ba8bfcd42 100644
--- a/Makefile
+++ b/Makefile
@@ -95,7 +95,7 @@ endif

 # emu for the release version
 RELEASE_ARGS += --disable-all --remove-assert --fpga-platform
-DEBUG_ARGS   += --enable-difftest
+DEBUG_ARGS   += --enable-difftest --enable-log
 ifeq ($(RELEASE),1)
 override SIM_ARGS += $(RELEASE_ARGS)
 else
```

YudeFang, 2023-11-25:
thank you，it really helps me

closed 2023-11-26 by Tang-Haojin