OpenXiangShan/XiangShan #2507: memBlock.io.ooo_to_mem.enqLsq 信号互联

MINUS-1ONE, 2023-11-26:
在XSCore.scala顶层互联代码实现中, 我发现exuBlocks.head和ctrlBlock都对memBlock.io.ooo_to_mem.enqLsq进行了互联:
https://github.com/OpenXiangShan/XiangShan/blob/51e45dbbf87325e45ff2af6ca86ed6c7eed04464/src/main/scala/xiangshan/XSCore.scala#L302-L322
并且从生成的Verilog来看, 前者应该是被后者覆盖了, 是否应该要把前者删掉呢? 我查看了[Pr#1632](https://github.com/OpenXiangShan/XiangShan/pull/1632/commits/037428499f6660b3e0aecc9d9593a3c197fc8f17#diff-978be84f9463c5afb827b7ca74250699509f0568247834eb8bf6be3fffbfd017R301-R321)是最早加入ctrlBlock.io.enqLsq <> memBlock.io.enqLsq的更改, 这里应该是让ctrlBlock负责向memBlock传输enqLsq信号了.

huxuan0307, 2023-11-28:
此处连线代码确实可以删除，感谢您的指正！由于Chisel的特性，这部分未用到的代码会被优化掉，所以没有功能上的错误。我们也注意到了这个问题，目前乱序后端（Backend）的重写接近完成，未来将会用新版代码替换掉这部分，所以暂时没有修改。目前开发的代码在`backend-231017`分支，未来会被合并到`master`分支中。

下面简单解释一下这部分代码用途（包括曾经的用途）。

在`enqLsq`这个bundle中，`canAccept`表示LoadStoreQueue（LSQ）当前是否可以接收新的访存指令，`req`表示有访存指令（非原子指令）需要进入LSQ，`needAlloc`用2bit区分指令进入LoadQueue（LQ）还是StoreQueue（SQ），`resp`用来传递最新的LQIDX和SQIDX。

访存指令的第二阶段dispatch模块（Dispatch2RS）在exuBlocks.head中，这个阶段的指令仍然是顺序的，需要同时完成将指令分派到ReservationStation（RS）和指令进入LSQ两个过程，所以需要来自LSQ的相关信息。这也是[XiangShan/src/main/scala/xiangshan/XSCore.scala](https://github.com/OpenXiangShan/XiangShan/blob/51e45dbbf87325e45ff2af6ca86ed6c7eed04464/src/main/scala/xiangshan/XSCore.scala#L302-L322) 这行连线的作用。但是MemBlock到ExuBlock的物理连线太长了，因为LSQ自身具备的顺序的特性，我们在ExuBlock内部的Scheduler里增加了`LsqEnqCtrl`模块用于同步LSQ的信息。有了LsqEnqCtrl后就不需要再从MemBlock连线，这部分IO没有实际用到，但是也没有被删除。

[translation]
This connection code can be removed. Thanks for the correction! Due to the features of Chisel, this unused part of the code is optimized so there are no functional bugs. We've also noticed this issue. The recent rewriting of the Backend is nearly completed, and the new code will replacing these code in the future. So no changes have been made for this. The new Backend code is currently developed in the `backend-231017` branch and will be merged into the `master` branch in a few weeks.

Here is a brief explanation of what this part of the code is used for (including what it was used for).

In the `enqLsq` bundle, `canAccept` indicates whether LoadStoreQueue (LSQ) can receive new instructions, `req` indicates that there are access instructions (non-atomic instructions) that need to enqueue LSQ, and `needAlloc` distinguishes with 2 bits whether an instruction enqueue the LoadQueue (LQ) or the StoreQueue (SQ), and `resp` is used to pass the latest LQIDX and SQIDX.

The second phase of the dispatch module (Dispatch2RS) for the load/store instructions is in exuBlocks.head. The instructions in this phase are still sequential, and the processes need to be done at the same time. The one is dispatching the instructions to the ReservationStation (RS), the other is enqueueing the instructions into the LSQ. So the relevant information from the LSQ is needed. This is the function of the connection at [XiangShan/src/main/scala/xiangshan/XSCore.scala](https://github.com/OpenXiangShan/XiangShan/blob/ 51e45dbbf87325e45ff2af6ca86ed6c7eed04464/src/main/scala/xiangshan/XSCore.scala#L302-L322). But the physical connection from MemBlock to ExuBlock is too long. Since the LSQ is organized in order, we add `LsqEnqCtrl` module to the Scheduler inside ExuBlock to synchronize the information of LSQ. With LsqEnqCtrl, there is no need to connect from MemBlock, this part of the IO is not actually used, but not removed.

yelbuod, 2023-11-28:
感谢您的详细解答!

closed 2023-11-29 by huxuan0307