OpenXiangShan/XiangShan #2510: a question about spec cpu 2006 in Nanhu

menglinhan, 2023-11-27:
您好，我这边最近在评估nanhu的spec cpu分数。有以下两个问题：
1） 根据目前我看到的公开的文档：https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/simpoint/，没有看到和spec cpu 相关的内容，比如程序。
2） 请问还有其他的和spec cpu评估相关的是公开的文档吗？比如在香山上跑通使用simpoint方法的spec cpu评估的指导手册。

eastonman, 2023-11-27:
The SimPoint sampling method and checkpoint have both been introduced in the document. The sampling assessment method is suitable for any program.
If you want to use this method to evaluate SPEC CPU, you only need to get the compiled binaries for each test point of SPEC CPU and then follow the steps described in the document. The source code for SPEC CPU cannot be made public due to copyright reasons.

simpoint采样方法和checkpoint都在文档中有介绍了，采样评估方法适用于任何程序。
如果你希望用这种方法评估SPEC CPU，只需要拿到编译好的SPEC CPU各测试点的二进制，然后按照文档描述的步骤即可。SPEC CPU的源码受版权原因限制是不能公开的。

menglinhan, 2023-11-28:
好的，多谢

menglinhan, 2023-11-28:
您好，请问你们有对其他riscv cpu做过评估吗？使用你们这种simpoint的方法。NEMU的tracing分支可以模拟其他riscv CPU并生成测试点信息吗？

Hello, have you evaluated other riscv cpus? Use your simpoint approach. Can the tracing branch of NEMU emulate other riscv cpus and generate test point information?

wakafa1, 2023-11-30:
我们没有对其他 riscv cpu 做过评估，但原理上该方法是通用的。

menglinhan, 2023-11-30:
好的，了解了，谢谢，我这边下载的NEMU是tracing分支的，

关于NEMU的使用有如下说明：
cpt-bk分支：在 NEMU 目录下执行 make ISA=riscv64 XIANGSHAN=1 生成 NEMU 的可执行文件。 tracing分支：在 NEMU 目录下执行 make riscv64-xs-simpoint_defconfig; make memuconfig # --> Save; make -j 生成 NEMU 的可执行文件。

执行“make memuconfig # --> Save”时会报如下log：
NEMU$ make memuconfig # --> Save
make: *** No rule to make target 'memuconfig'.  Stop.

请问是什么原因

cebarobot, 2023-11-30:
> 好的，了解了，谢谢，我这边下载的NEMU是tracing分支的，
>
> 关于NEMU的使用有如下说明： cpt-bk分支：在 NEMU 目录下执行 make ISA=riscv64 XIANGSHAN=1 生成 NEMU 的可执行文件。 tracing分支：在 NEMU 目录下执行 make riscv64-xs-simpoint_defconfig; make memuconfig # --> [...]

拼写错误，应该是 `make menuconfig`，我们稍后修正一下文档。
It'a typo. The correct command is `make menuconfig`. We will correct it shortly.

menglinhan, 2023-11-30:
好的，了解了，多谢

closed 2023-12-13 by Tang-Haojin