OpenXiangShan/XiangShan #2541: Missing 4-Byte Alignment Check for JALR Instruction

youzi27, 2023-12-09:
**Describe the bug**
The XiangShan implementation of the JALR instruction appears to not perform the required 4-byte alignment check as specified in the RISC-V ISA specification Section 2.5:

> The JAL and JALR instructions will generate an instruction-address-misaligned exception if the target address is not aligned to a four-byte boundary.

The specification states that the JAL and JALR instructions should generate an instruction-address-misaligned exception if the target address is not aligned to a **four-byte boundary**. However, this behavior **is not** observed in the current implementation.

**To Reproduce**
_Please refer to the attached detailed execution file and logs._
```
- ...
- li x15,0x00000000800a100b
- jalr x15
- ...
```

**Expected behavior**
When executing the JALR instruction with a target address that is not 4-byte aligned, an **instruction-address-misaligned** exception should be raised, as per the RISC-V ISA specification.

**Screenshots**

- The execution of the JALR instruction does not trigger any exceptions (even if the target address is not 4-byte aligned), but instead directly attempts to jump to the unaligned target address. This behavior **does not** comply with the specification.

[image]

The exception in the screenshot is caused by an exception generated when accessing data at the jump target address. Due to the unaligned address, decoding fails, resulting in an **illegal instruction exception**.

**Environment (optional, if necessary):**
 - OS: Ubuntu 22.04.3 LTS
 - Compiler: gcc 11.4.0
 - XiangShan: * master branch (commit 3b1a683bf8b2a904a7e4d56372b2cec6ba0ae66c)

**Additional context**
XiangShan seems to only clear the address's least significant bit for the JALR instruction, as specified in the standard: 

> Clearing the least-significant bit when calculating the JALR target address both simplifies the hardware slightly and allows the low bit of function pointers to be used to store auxiliary information.

 However, there is no 4-byte alignment check for the JALR target address.

**Testcase**
[test.zip](https://github.com/OpenXiangShan/XiangShan/files/13623112/test.zip)

chenguokai, 2023-12-10:
It is not a bug. When RVC is enabled, jalr can jump to a address that is not 4-bytes aligned.

closed 2023-12-10 by chenguokai

youzi27, 2023-12-11:
> It is not a bug. When RVC is enabled, jalr can jump to a address that is not 4-bytes aligned.
> 
> Check page 16 of risc-v spec.

Hi @chenguokai  , thank you very much for your explanation. 
By the way, does Xiangshan support transitioning to U mode after mret?

chenguokai, 2023-12-11:
> By the way, does Xiangshan support transitioning to U mode after mret?

It should work according to the spec. If you encounter any problem, feel free to open another issue.