OpenXiangShan/XiangShan #2543: question about benchmark

menglinhan, 2023-12-12:
您好，请问你们在南湖上有跑过linpack和whetstone这两个benchmark吗？我在跑自己找的benchmark时，发现缺少一些函数（例如fabs），不知道你们是怎么解决的？

Hello！have you run the benchmark of linpack and whetstone in Nanhu? when I run these two benchmarks which found from opensource, some functions were missing. How did you solve this problem？

menglinhan, 2023-12-13:
或者说，需要include哪个头文件才能包括相关的数学函数库？比如sin和cos，log函数

cebarobot, 2023-12-13:
我们没有在香山上运行过 linpack 或者 whetstone。有关 linpack 或者 whetstone 编译运行相关的问题是纯软件问题，这与香山处理器关系不大，需要您自行解决。

We have not run Linpack or Whetstone on Mount Hua. Issues related to compiling and running Linpack or Whetstone are purely software-related and are not significantly related to the Mount Hua processor. You will need to resolve these matters on your own.

menglinhan, 2023-12-13:
好的，多谢。
再咨询一下，香山官方提供的 gcc-riscv64-linux-gnu编译器看起来是不能识别cos，log这些函数的，麻烦确认一下，不知道是我使用的问题，还是说原本就不支持这些数学函数。

cebarobot, 2023-12-13:
首先，`riscv64-linux-gnu-gcc` 编译器是由 gnu 提供的 RISC-V 工具链的一部分。它不是由香山提供的。
其次，要在 C 语言中使用 `cos()`、`log()` 等数学函数并用 GCC 编译（无论是 x86 还是 RISC-V 平台），您需要：
* 引用 `math.h` 头文件，例如 `#include <math.h>`；
* 在 gcc 编译命令中添加 `-lm` 参数，例如 `riscv64-linux-gnu-gcc test.c -lm`。

Firstly, the `riscv64-linux-gnu-gcc` compiler is part of the GNU RISC-V toolchain, NOT provided by XiangShan.
Secondly, to use mathematical functions like `cos()` and `log()` in the C language and compile with GCC (whether on x86 or RISC-V platforms), you need to:
* Include the math.h header file, for example: `#include <math.h>`;
* Add the `-lm` parameter to the gcc compile command, for instance: `riscv64-linux-gnu-gcc test.c -lm`.

menglinhan, 2023-12-14:
我们是这样做的，但还是无法识别：

cebarobot, 2023-12-14:
请提供您编译的源代码文件和报错信息。
Please provide the source code files you tried to compile and the error messages.

menglinhan, 2023-12-15:
报错信息如下：
riscv64-linux-gnu-ld: ****/xs-env-master/nexus-am/apps/whetstone/whetstone.c:239: undefined reference to `sin'
riscv64-linux-gnu-ld: ****/xs-env-master/nexus-am/apps/whetstone/whetstone.c:240: undefined reference to `cos'
riscv64-linux-gnu-ld: ****/xs-env-master/nexus-am/apps/whetstone/whetstone.c:240: undefined reference to `cos'

使用的源代码在这个位置：https://github.com/mollybuild/RISCV-Measurement/blob/master/Benchmarks/Whetstone/whetstone.c

cebarobot, 2023-12-15:
我不确定您是如何使用 nexus-am 编译 whetstone 的，请提供更多有关您如何移植 whetstone 至 nexus-am 框架的信息。
但值得一提的是，nexus-am 在通过 make 生成内存镜像 bin 文件时，编译和链接是分开完成的。您可能需要在 `LDFLAGS` 或相关参数中添加 `-lm`。

**[TRANSLATION]** I'm not sure how you compile Whetstone using nexus-am. Please provide more information on how you ported Whetstone to the nexus-am framework.
However, it's worth noting that in nexus-am, when generating a memory image bin file through 'make', compilation and linking are performed separately. You might need to add `-lm` in `LDFLAGS` or related parameters during the process.

chenguokai, 2023-12-15:
建议您使用工具链提供的gcc直接编译含数学函数的最小测试用例，如果`gcc -lm`能够正常编译，那么工具链没有问题。应用程序的移植问题是另一个独立的问题，我们无法提供相关支持，您需要自行解决。

You should try compile minimal test case with math functions using gcc provided by the toolchain. If `gcc -lm` compiles well, the toolchain should be fine. Porting applications to nexus-am is another issue that should be resolved yourself.

menglinhan, 2023-12-15:
关于LDFLAGS后加-lm，编译出现问题。
好的，那这些问题我自己这边会继续看的，多谢你们的支持。
最后我想问一下，你们有使用过cos，log这些函数用指定的编译器编译吗？

cebarobot, 2023-12-15:
我们有用过 riscv64 的工具链编译 cos、log 等数学函数，他们能正常工作。
We have used the riscv64 toolchain to compile mathematical functions like cos and log, and they work properly.

menglinhan, 2023-12-15:
好的，多谢

closed 2023-12-15 by cebarobot

menglinhan, 2023-12-15:
方便给一下你们安装的riscv64的开源链接吗？我也安装一个相同的

cebarobot, 2023-12-15:
> 方便给一下你们安装的riscv64的开源链接吗？我也安装一个相同的 [TRANSLATION] Could you please provide the open-source link for the RISC-V64 toolchain that you've installed? I'd like to install the same one.

请参考：https://xiangshan-doc.readthedocs.io/zh-cn/latest/compiler/gnu_toolchain/
Please refer to https://xiangshan-doc.readthedocs.io/zh-cn/latest/compiler/gnu_toolchain/

请注意，**RISC-V 工具链不由香山提供，香山也无法提供任何支持**。
Please note that **the RISC-V toolchain is not provided by XiangShan, and XiangShan cannot offer any support regarding it**.

menglinhan, 2023-12-15:
好的，多谢支持