OpenXiangShan/XiangShan #2544: Unexpected Behavior When Executing 'mret' Instruction

youzi27, 2023-12-12:
**Describe the bug**
After setting the correct MEPC and MATATUS registers and disabling SATP, when I execute the MRET instruction in M mode to return to U mode, an 'Instruction access fault' exception occurs unexpectedly.

**To Reproduce**
```
...
mepc_setup:       
                  la x19, test
                  csrw 0x341, x19
u_mode_setup:
                  li x14, 0x0
                  csrw 0x180, x14 # SATP
                  li x19, 0xa00140000 # return u mode
                  csrw 0x300, x19 # MSTATUS
                  li x19, 0x0
                  csrw 0x304, x19 # MIE
                  mret
test:
                  addi x1, x2, 2 # test
                  nop
...
```
The above instruction generates an 'Instruction access fault' exception after executing the MRET instruction.

**Expected behavior**
After executing MRET, the mode should switch, and execution should proceed normally at the MEPC with the correct settings.

**Screenshots**
[image]

**Environment (optional, if necessary):**

- OS: Ubuntu 22.04.3 LTS
- Compiler: gcc 11.4.0
- XiangShan: * master branch (commit https://github.com/OpenXiangShan/XiangShan/commit/3b1a683bf8b2a904a7e4d56372b2cec6ba0ae66c)

**Additional**
I executed the same command in Rocket Chip and found that it could successfully mret without any exceptions. So I don't know why this is happening in the XiangShan. If I missed any details in the XiangShan design, please let me know. Thank you very much!

chenguokai, 2023-12-14:
May you provide a minimal working example on this mismatch?

chenguokai, 2023-12-14:
You should setup PMP before you jump to U mode. Check risc-v priviledge spec:
[image]

chenguokai, 2023-12-14:
Please check risc-v spec **carefully** if you are to report any bug.

closed 2023-12-14 by chenguokai

youzi27, 2023-12-14:
> You should setup PMP before you jump to U mode. Check risc-v priviledge spec: <img alt="image" width="823" src="https://private-user-images.githubusercontent.com/11240171/290395746-4e87ecf1-688f-4feb-b236-9ab661965ea0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1MjY3NTYsIm5iZiI6MTcwMjUyNjQ1NiwicGF0aCI6Ii8xMTI0MDE3MS8yOTAzOTU3NDYtNGU4N2VjZjEtNjg4Zi00ZmViLWIyMzYtOWFiNjYxOTY1ZWEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDA0MDA1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0ZDg4NWQ1YjQyYzcyNTEyZDcxMjFhNzhhYWUxOTgwZGM3NTcxNWYxNzYyNzM0ZDM5N2QwYTI5NDUzMTdlOWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xiMCzUrg7LnsJDdZGTBlckHHHc9JUssYLEuKSKTiANI">

Hi @chenguokai 
Thank you for your kind reminder, I will reexamine my code.