OpenXiangShan/XiangShan #2545: 构建 BBL 并链接 Kernel 出现问题

YudeFang, 2023-12-13:
**Describe the bug**
构建 BBL 并链接 Kernel时，出现编译错误
好像是不支持zicsr这个扩展？我整个步骤是按照香山官方文档中的gnu-toolchain来配置的 gnu版本也是通过 curl https://mirror.iscas.ac.cn/riscv-toolchains/git/riscv-collab/riscv-gnu-toolchain.sh | bash 这条命令进行配置

**Environment (optional, if necessary):**
 - OS: [Ubuntu 20.04 LTS]
 - Compiler: [GCC 9.4.0]
 - 
**Screenshots**
[image]

**Additional context**

chenguokai, 2023-12-13:
这是上游gcc更新加入的约束，老版本的BBL和Linux没有相关的处理，因此会出错，可参考[相关讨论](https://github.com/efabless/caravel_board/issues/37)。您可以考虑使用老版本编译器来编译BBL/Linux。

Upstream gcc added zicsr constraint which old BBL and Linux repos did not handle. You may refer to [Related discussion](https://github.com/efabless/caravel_board/issues/37) for more detail. You may use older compiler for BBL/Linux compilation.

YudeFang, 2023-12-13:
请问之前测试用的gcc版本是？

cebarobot, 2023-12-13:
我们使用的应该是 riscv64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1\~20.04) 9.4.0
We used riscv64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1\~20.04) 9.4.0

cebarobot, 2023-12-18:
Please add `_zicsr_zifence` at `riscv-linux/arch/riscv/Makefile` like this.
```diff
--- a/arch/riscv/Makefile
+++ b/arch/riscv/Makefile
@@ -50,9 +50,9 @@ ifeq ($(CONFIG_RISCV_ISA_C),y)
        KBUILD_ARCH_C = c
 endif

-KBUILD_AFLAGS += -march=$(KBUILD_MARCH)$(KBUILD_ARCH_A)fd$(KBUILD_ARCH_C)
+KBUILD_AFLAGS += -march=$(KBUILD_MARCH)$(KBUILD_ARCH_A)fd$(KBUILD_ARCH_C)_zicsr_zifencei

-KBUILD_CFLAGS += -march=$(KBUILD_MARCH)$(KBUILD_ARCH_A)$(KBUILD_ARCH_C)
+KBUILD_CFLAGS += -march=$(KBUILD_MARCH)$(KBUILD_ARCH_A)$(KBUILD_ARCH_C)_zicsr_zifencei
 KBUILD_CFLAGS += -mno-save-restore
 KBUILD_CFLAGS += -DCONFIG_PAGE_OFFSET=$(CONFIG_PAGE_OFFSET)
```

And add `_zicsr_zifence` at `riscv-pk/Makefile` like this.
```diff
--- a/Makefile
+++ b/Makefile
@@ -22,7 +22,7 @@ BBL_BIN = $(BBL_BUILD_PATH)/bbl.bin
 BBL_PAYLOAD = $(LINUX_ELF)
 #BBL_PAYLOAD = dummy_payload
 BBL_CONFIG = --host=riscv64-unknown-elf --with-payload=$(BBL_PAYLOAD) \
-                                                --with-arch=rv64imac --enable-logo #--enable-print-device-tree
+                                                --with-arch=rv64imac_zicsr_zifencei --enable-logo #--enable-print-device-tree

 DTB = $(BBL_BUILD_PATH)/system.dtb
 DTS = dts/system.dts
```

This should resolve the issue encountered when compiling older code with a new compiler.

closed 2024-01-05 by cebarobot