OpenXiangShan/XiangShan #2548: Unexpected Behavior in LR/SC Sequence with Branch Prediction Errors

youzi27, 2023-12-13:
**Describe the bug**
A deviation from the expected behavior is observed in the LR/SC sequence in RISC-V architecture when branch prediction is involved. Specifically, when a branch prediction error occurs, the incorrectly predicted instruction seems to influence the success of subsequent SC instruction executions.

**To Reproduce**

```
...
test:             la x1, test_1
                  la x11, test_1+1 # to test prediction failures
                  addi x2, x0, 0
                  addi x3, x0, 3
                  li x4, 0
                  li x5, 7             # loop times | train branch prediction
                  lr.d x10, (x28)
loop:             addi  x4, x4, 1     
                  bge   x4, x5, done
                  j loop
                  sc.d x3, x2, (x11) #  to test prediction failures
done:             sc.d x3, x2, (x1)
                  nop
...
```
**Expected behavior**
The expected behavior is that the success of the SC instruction should depend solely on the state of the memory address reserved by the LR instruction, and should not be influenced by branch prediction errors or the execution of unpredicted instructions.

**Environment (optional, if necessary):**
- OS: Ubuntu 22.04.3 LTS
- Compiler: gcc 11.4.0

**Additional context**
When the loop count is greater than or equal to 7, SC fails to execute. When it's less than 7, SC executes successfully. If I missed any details in the XiangShan design, please let me know. Thank you very much!

cebarobot, 2023-12-13:
Please review Chapter 8.3 of the RISC-V Spec, which delineates the essential properties that lead to the eventual success of Store-Conditional instructions.

youzi27, 2023-12-14:
Hi @cebarobot 
Thank you very much for your response. You are correct, **but** this still indirectly suggests that there may be some incorrect handling by Xiangshan when it comes to branch prediction. To demonstrate this issue, let's reconstruct the assembly as follows:

```
...
test:             la x1, test_1
                  la x11, test_1+111 # to test prediction failures
                  addi x2, x0, 0
                  addi x3, x0, 3
                  li x4, 0
                  li x5, 7             # loop times | train branch prediction
                  lr.d x10, (x28)
loop:             addi  x4, x4, 1     
                  bge   x4, x5, done
                  j loop
                  la x11, test_1#  to test prediction failures
done:             lr.d x10, (x11) # the correct value for x11 is test_1+ 111. An exception: Load address misaligned
                  sc.d x3, x2, (x11) # Under normal circumstances, SC never succeeds.
                  nop
...
```
Because the address loaded by x11 is not aligned, it triggers an exception (exception code 04). However, due to branch prediction executing 'la x11, test_1,' it erroneously affects the 'lr.d x10, (x11)' instruction. As a result, the final 'sc' **succeeds**."

Tang-Haojin, 2023-12-14:
Could you please attach the full binary image which can reproduce your issue?

youzi27, 2023-12-14:
> Could you please attach the full binary image which can reproduce your issue?

Hi @Tang-Haojin 
Thank you very much for your response. After remaking the XiangShan environment, I unexpectedly found that this issue disappeared, and I'm not sure why. If I figure it out, I will reopen the question. So, I will close this issue for now. Thanks again.