OpenXiangShan/XiangShan #2551: about address map

menglinhan, 2023-12-14:
您好，我们遇到一个关于访问地址的问题想咨询一下你们团队。
在我们对地址0x00000800写数据时，cpu会挂死在这里，我们将代码放在0x10000000起始的flash空间中。

chenguokai, 2023-12-14:
硬件不保证对地址空洞的访问能够得到妥善处理，请参考香山编译时生成的设备树确定可访问地址范围。

menglinhan, 2023-12-14:
请问哪里能查看香山的合法访问范围？设备树哪里能看到？

cebarobot, 2023-12-14:
设备树文件位于 `build/dts`，里面包含了合法的内存地址范围信息。是的，是内部进行判断的。

Device tree file is at `build/dts`, which contains accessible address ranges. Yes, it is determined internally.

menglinhan, 2023-12-14:
是这个文件 吗？xs-env/XiangShan/build/XSTop.dts

cebarobot, 2023-12-14:
> 是这个文件 吗？xs-env/XiangShan/build/XSTop.dts
> 
> **[TRANSLATION]** Is this the file? `xs-env/XiangShan/build/XSTop.dts`

看起来是的。
It seems it is.

menglinhan, 2023-12-14:
请问有资料介绍设备树怎么看吗？直接看该文件看不太明白

cebarobot, 2023-12-14:
> 请问有资料介绍设备树怎么看吗？直接看该文件看不太明白 **[TARNSLATION]** Is there any documentation available that explains how to interpret device trees? I find it difficult to understand the file by directly looking at it.

请在互联网上搜索相关资料。Please Search The Friendly Web.

menglinhan, 2023-12-14:
好的，多谢。我去了解一下

closed 2023-12-21 by chenguokai