OpenXiangShan/XiangShan #2561: L2Prefetch parameter Uninitialized

xiaokamikami, 2023-12-18:
The L2Prefetch parameter string is uninitialized, which causes a compiler error on the hardware accelerated emulator.

The code generated by constantin attempts to assign the value of DPIC to a variable of class wire, which is not allowed on some platforms.If you want to use this syntax, you should replace the wire of the code template with a trigger.

L2预取器中的parameter string 未初始化，会引起在硬件加速仿真器上的编译报错。
constantin功能生成的代码尝试将DPIC的值赋值给wire类变量，这在某些平台上是不允许的语法，可以把代码模板的wire换为触发器。

shinezyy, 2023-12-18:
Can you please the code site that assigning to wire?

xiaokamikami, 2023-12-18:
For example
[image]
Cause an error
[image]

Tang-Haojin, 2023-12-22:
> The L2Prefetch parameter string is uninitialized, which causes a compiler error on the hardware accelerated emulator.

Could you please provide the error site and message of this problem?

xiaokamikami, 2023-12-22:
> > The L2Prefetch parameter string is uninitialized, which causes a compiler error on the hardware accelerated emulator.
> 
> Could you please provide the error site and message of this problem?

[image]

after I pulled the repository, it was fixed, thanks

closed 2023-12-22 by xiaokamikami