OpenXiangShan/XiangShan #2602: 使用虚拟机编译FPGA原型验证版本出现jave heap space 问题

ShmilyM-X, 2023-12-29:
====== Non-inclusive L3 (1.5MB * 4-bank) prefetch: None ======
bankBits: 2
sets:4096 ways:6 blockBytes:64
[client] size:1.0MB
[client] sets:2048 ways:8 blockBytes:64
blockGranularityBits: 9
usr/preferCache: (1-bit)
echo/blockisdirty: (1-bit)
clientBits: 2
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at java.base/java.lang.Class.getDeclaredMethods0(Native Method)
	at java.base/java.lang.Class.privateGetDeclaredMethods(Class.java:3166)
	at java.base/java.lang.Class.privateGetPublicMethods(Class.java:3191)
	at java.base/java.lang.Class.getMethods(Class.java:1904)
	at chisel3.internal.HasId.getPublicFields(Builder.scala:309)
	at chisel3.internal.HasId.getPublicFields$(Builder.scala:292)
	at chisel3.experimental.BaseModule.getPublicFields(Module.scala:382)
	at chisel3.experimental.BaseModule.nameIds(Module.scala:607)
	at chisel3.RawModule.generateComponent(RawModule.scala:71)
	at chisel3.Module$.do_apply(Module.scala:71)
	at huancun.Slice.$anonfun$ms$2(Slice.scala:94)
	at huancun.Slice$$Lambda$38345/0x0000000843f8cc40.apply(Unknown Source)
	at scala.collection.generic.GenTraversableFactory.fill(GenTraversableFactory.scala:92)
	at huancun.Slice.$anonfun$ms$1(Slice.scala:92)
	at huancun.Slice$$Lambda$38344/0x0000000843f8c840.apply(Unknown Source)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at huancun.Slice.<init>(Slice.scala:91)
	at huancun.HuanCun$$anon$1.$anonfun$slices$7(HuanCun.scala:325)
	at huancun.HuanCun$$anon$1$$Lambda$37185/0x0000000843e05c40.apply(Unknown Source)
	at chisel3.Module$.do_apply(Module.scala:53)
	at huancun.HuanCun$$anon$1.$anonfun$slices$6(HuanCun.scala:325)
	at huancun.HuanCun$$anon$1$$Lambda$37184/0x0000000843e05840.apply(Unknown Source)
	at chisel3.withReset$.apply(MultiClock.scala:66)
	at huancun.HuanCun$$anon$1.$anonfun$slices$5(HuanCun.scala:325)
	at huancun.HuanCun$$anon$1$$Lambda$37183/0x0000000843e05440.apply(Unknown Source)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at huancun.HuanCun$$anon$1.$anonfun$slices$2(HuanCun.scala:325)
	at huancun.HuanCun$$anon$1$$Lambda$37180/0x0000000843e04840.apply(Unknown Source)
	at scala.collection.immutable.List.map(List.scala:293)
	at huancun.HuanCun$$anon$1.$anonfun$slices$1(HuanCun.scala:319)
	at huancun.HuanCun$$anon$1$$Lambda$37179/0x0000000843e03c40.apply(Unknown Source)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
1 targets failed
XiangShan.runMain subprocess failed
make: *** [build/XSTop.v] Error 1

LOG信息如上，
使用Centos7 系统，内存56G，尝试编译2core失败，编译1core失败
调整build.sc中-Xmx参数为10G失败，
尝试增加环境变量：
_JAVA_OPTIONS=-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K 后编译依然失败
是否有更好的办法进行定位或解决该问题呢？

cebarobot, 2023-12-29:
请考虑使用更大的内存和更大的 swap 空间。`-Xmx` 参数的值也可以考虑继续增大。

Please consider using larger memory and increasing the swap space. You might also want to consider further increasing the value of the -Xmx parameter.

Tang-Haojin, 2024-03-15:
我们应该已经解决了您的问题。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

Your issue may have been solved. If you have any other questions, please feel free to open the issue again or file a new one.

closed 2024-03-15 by Tang-Haojin