OpenXiangShan/XiangShan #2606: issue about kunminghu

menglinhan, 2024-01-02:
您好，咨询一下，我这边上周五从git上拉了最新的kunminghu代码，自己生成verilog后，搭建环境用vcs仿真，io_riscv_rst_vec_0设置的是38'h10000000，我理解应该是和南湖一样从peripheral口有取指访问发出，但peripheral和memory口均没有访问发出，请问这个正常吗？可能的原因是什么？方便提供一下pc路径吗？谢谢

wakafa1, 2024-01-03:
你好，这是预期之外的行为，请尝试参考一下 https://github.com/OpenXiangShan/XiangShan/blob/master/src/test/scala/top/SimTop.scala 中的代码对 XSTop 顶层 IO 进行赋值；如果仍然不起作用，请关注 https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/frontend/icache/InstrUncache.scala 模块的信号判断是否收到取指单元的第一条取指请求。

Hi, this is unexpected behavior. Please try to assign XSTop top-level IO by referring to the code in https://github.com/OpenXiangShan/XiangShan/blob/master/src/test/scala/top/SimTop.scala; if it still doesn’t work, please pay attention to https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/frontend/icache/InstrUncache.scala module’s signal to determine whether it has received the first instruction fetch request.

menglinhan, 2024-01-04:
IO的赋值是参考的SimTop中的值。InstrUncache模块接口的input没有看到请求进来。有可能是编译的warning导致的吗？编译的时候有几个位宽相关的warning，下面是第一个。

List(xiangshan.backend.fu.Jump@ff75c80, xiangshan.backend.fu.CSR@1890648d, xiangshan.backend.fu.Fence@10f1d99, xiangshan.backend.fu.fpu.IntToFP@6e9e11ff) List(xiangshan.backend.fu.Jump@ff75c80) hasRedirect: 1
[warn] src/main/scala/xiangshan/backend/fu/CSR.scala:1091:21: Dynamic index with width 4 is too small for extractee of width 64
[warn]   val delegS = deleg(causeNO(3,0)) && (priviledgeMode < ModeM)
[warn]                     ^
[warn] There were 1 warning(s) during hardware elaboration.

mill的话，我尝试过0.9.8和0.9.9，这两个版本可以吗？

menglinhan, 2024-01-05:
您好，仿真出问题的原因找到了，是因为没有define RANDOMIZE_REG_INIT。多谢了

wakafa1, 2024-01-05:
> 您好，仿真出问题的原因找到了，是因为没有define RANDOMIZE_REG_INIT。多谢了。 在之前的nanhu中同样没有define这个宏，应该是在kunminghu中寄存器的x态影响到了功能。

感谢你的反馈，我们之后会尝试使用 VCS 进行一下仿真确认 X 态传播的问题