OpenXiangShan/XiangShan #2636: why the idx for psrc2 here be 1 in renaming？

gxflying, 2024-01-12:
hi， see blow,  should the psrc(1)  be psrc(2) in the line 178?
https://github.com/OpenXiangShan/XiangShan/blob/8fae59bba57fd80fcd1d85aadbf87895b97d167a/src/main/scala/xiangshan/backend/rename/Rename.scala#L178

```
    // int psrc2 should be bypassed from next instruction if it is fused
    if (i < RenameWidth - 1) {
      when (io.fusionInfo(i).rs2FromRs2 || io.fusionInfo(i).rs2FromRs1) {
        uops(i).psrc(1) := Mux(io.fusionInfo(i).rs2FromRs2, io.intReadPorts(i + 1)(1), io.intReadPorts(i + 1)(0))
      }.elsewhen(io.fusionInfo(i).rs2FromZero) {
        uops(i).psrc(1) := 0.U
      }
    }
```

poemonsense, 2024-01-12:
RISC-V instructions generally have two operands called `rs1` and `rs2`. However, in most programming languages, indices start with 0. So `psrc(1)` corresponds to `rs2`.

closed 2024-01-12 by poemonsense

gxflying, 2024-01-12:
year， But,  I think the code snippet above is for fusion instructions who will has 3 src reg
and accroding to the comment ,  the assignment is for the 3rd reg,    doesn't it ?

poemonsense, 2024-01-12:
It's for rs2 as shown in the code you cited.

cebarobot, 2024-01-12:
The code you cited is full of `rs2`, such as `rs2FromRs2`, `rs2FromRs1`. It is obviously about the second reg.