OpenXiangShan/XiangShan #2638: VCS Simulatrion Run Err

wyz-icer, 2024-01-12:
使用VCS进行仿真时报错如下： 
（log err 贴在环境配置下）

环境配置如下：
master版本：
~/xs-env.kunminghu.master/XiangShan$ git log -1
commit 8fae59bba57fd80fcd1d85aadbf87895b97d167a (HEAD -> master, origin/master, origin/HEAD)
Author: Easton Man <me@eastonman.com>
Date:   Mon Jan 8 11:12:15 2024 +0800
:~/xs-env.kunminghu.master/XiangShan/difftest$ git log -1
commit 9ac68320d50615441785a7b6f2cd61773403651f (HEAD)
Author: Yinan Xu <xuyinan@ict.ac.cn>
Date:   Sat Dec 23 11:03:20 2023 +0800

    difftest: fix the init status of diffstate_buffer (#229)
vcs版本：
/home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/bin/vcs

GCC版本：
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0

运行命令： 
~/xs-env.kunminghu.master/XiangShan$ make simv RELEASE=1
 
log err Info： 
if [ -x /home/cp1/xs-env.kunminghu.master/XiangShan/build/simv ]; then chmod a-x /home/cp1/xs-env.kunminghu.master/XiangShan/build/simv; fi
g++  -o /home/cp1/xs-env.kunminghu.master/XiangShan/build/simv   -no-pie    -lsqlite3 -lz -lpthread -ldl -rdynamic  -Wl,-rpath='$ORIGIN'/simv.daidir -Wl,-rpath=./simv.daidir --rpath=/home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib -L/home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib  -Wl,-rpath-link=./  sdcard.o golden.o ram.omJTAG.o common.o uart.o vga.o keyboard.o flash.o remote_bitbang.o coverage.o lightsss.o device.o compress.o dut.o main.o difftest-dpic.o difftrace.o refproxy.o difftest.o goldem.o chisel_db.o constantin.o spikedasm.o vcs_main.o   objs/a/amcQw_d.o   _355240_archive_1.so _355273_archive_1.so _355274_archive_1.so _355275_archive_1.so _355276_archive_1 _355277_archive_1.so _355278_archive_1.so _355279_archive_1.so _355280_archive_1.so _355281_archive_1.so _355282_archive_1.so _355283_archive_1.so _355284_archive_1.so _35528rchive_1.so _355286_archive_1.so _355287_archive_1.so _355288_archive_1.so _355289_archive_1.so _355290_archive_1.so _355291_archive_1.so _355292_archive_1.so _355293_archive_o _355294_archive_1.so _355295_archive_1.so _355296_archive_1.so _355297_archive_1.so _355298_archive_1.so _355299_archive_1.so _355300_archive_1.so _355301_archive_1.so _3553archive_1.so _355303_archive_1.so _355304_archive_1.so _355305_archive_1.so _355306_archive_1.so _355307_archive_1.so _355308_archive_1.so _355309_archive_1.so _355310_archiveso _355311_archive_1.so _355312_archive_1.so _355313_archive_1.so _355314_archive_1.so _355315_archive_1.so _355316_archive_1.so _355317_archive_1.so _355318_archive_1.so _355_archive_1.so _355320_archive_1.so _355321_archive_1.so _355322_archive_1.so _355323_archive_1.so _355324_archive_1.so _355325_archive_1.so _355326_archive_1.so _355327_archiv.so _355328_archive_1.so _355329_archive_1.so _355330_archive_1.so _355331_archive_1.so _355332_archive_1.so _355333_archive_1.so _355334_archive_1.so _355335_archive_1.so _356_archive_1.so _355337_archive_1.so _355338_archive_1.so _355339_archive_1.so _355340_archive_1.so _355341_archive_1.so _355342_archive_1.so _355343_archive_1.so _355344_archi1.so _355345_archive_1.so _355346_archive_1.so _355347_archive_1.so _355348_archive_1.so _355349_archive_1.so _355350_archive_1.so _355351_archive_1.so _355352_archive_1.so _353_archive_1.so _355354_archive_1.so _355355_archive_1.so _355356_archive_1.so _355357_archive_1.so _355358_archive_1.so _355359_archive_1.so _355360_archive_1.so _355361_arch_1.so _355362_archive_1.so _355363_archive_1.so _355364_archive_1.so _355365_archive_1.so _355366_archive_1.so _355367_archive_1.so _355368_archive_1.so _355369_archive_1.so _370_archive_1.so _355371_archive_1.so _355372_archive_1.so _355373_archive_1.so _355374_archive_1.so _355375_archive_1.so _355376_archive_1.so _355377_archive_1.so _355378_arce_1.so _355379_archive_1.so _355380_archive_1.so _355381_archive_1.so _355382_archive_1.so _355383_archive_1.so _355384_archive_1.so _355385_archive_1.so _355386_archive_1.so 5387_archive_1.so _355388_archive_1.so _355389_archive_1.so _355390_archive_1.so _355391_archive_1.so _355392_archive_1.so _355393_archive_1.so _355394_archive_1.so _355395_arve_1.so _355396_archive_1.so _355397_archive_1.so _355398_archive_1.so _355399_archive_1.so _355400_archive_1.so _355401_archive_1.so _355402_archive_1.so _355403_archive_1.so55404_archive_1.so _355405_archive_1.so _355406_archive_1.so _355407_archive_1.so _355408_archive_1.so _355409_archive_1.so _355410_archive_1.so _355411_archive_1.so _355412_aive_1.so _355413_archive_1.so _355414_archive_1.so _355415_archive_1.so _355416_archive_1.so _355417_archive_1.so _355418_archive_1.so _355419_archive_1.so _355420_archive_1.s355421_archive_1.so _355422_archive_1.so _355423_archive_1.so _355424_archive_1.so _355425_archive_1.so _355426_archive_1.so _355427_archive_1.so _355428_archive_1.so _355429_hive_1.so _355430_archive_1.so _355431_archive_1.so _355432_archive_1.so _355433_archive_1.so _355434_archive_1.so _355435_archive_1.so _355436_archive_1.so _355437_archive_1._355438_archive_1.so _355439_archive_1.so _355440_archive_1.so _355441_archive_1.so _355442_archive_1.so _355443_archive_1.so _355444_archive_1.so _355445_archive_1.so _355446chive_1.so _355447_archive_1.so _355448_archive_1.so  SIM_l.o      rmapats_mop.o rmapats.o rmar.o rmar_nd.o  rmar_llvm_0_1.o rmar_llvm_0_0.o          -lnuma -lvirsim -lerrorinlsnpsmalloc -lvfs    -lvcsnew -lsimprofile -luclinative /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/vcs_tls.o   -Wl,-whole-archive  -lvcsucli    -Wl,-no-wholechive       ./../simv.daidir/vc_hdrs.o  _vcs_pli_stub_.o   /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/vcs_save_restore_new.o /home/cp1/tools/synopsys/verdi/vi/R-2020.12-SP1/share/PLI/VCS/LINUX64/pli.a -ldl  -lc -lm -lpthread -ldl
**/usr/bin/ld: compress.o: in function `snapshot_compressToFile(unsigned char*, char const*, long)':
compress.cpp:(.text+0xa2): undefined reference to `gzopen'**
/usr/bin/ld: compress.cpp:(.text+0x114): undefined reference to `gzwrite'
/usr/bin/ld: compress.cpp:(.text+0x161): undefined reference to `gzclose'
/usr/bin/ld: compress.o: in function `readFromGz(void*, char const*, long, unsigned char)':
compress.cpp:(.text+0x1d5): undefined reference to `gzopen'
/usr/bin/ld: compress.cpp:(.text+0x2bf): undefined reference to `gzread'
/usr/bin/ld: compress.cpp:(.text+0x2ea): undefined reference to `gzread'
/usr/bin/ld: compress.cpp:(.text+0x2fe): undefined reference to `gzclose'
/usr/bin/ld: chisel_db.o: in function `init_db_TLLog()':
chisel_db.cpp:(.text+0xc6): undefined reference to `sqlite3_exec'
/usr/bin/ld: chisel_db.o: in function `TLLog_write':
chisel_db.cpp:(.text+0x2b0): undefined reference to `sqlite3_exec'
/usr/bin/ld: chisel_db.o: in function `init_db(bool, bool, char const*)':
chisel_db.cpp:(.text+0x36c): undefined reference to `sqlite3_open'
/usr/bin/ld: chisel_db.cpp:(.text+0x38a): undefined reference to `sqlite3_errmsg'
/usr/bin/ld: chisel_db.o: in function `save_db(char const*)':
chisel_db.cpp:(.text+0x1032): undefined reference to `sqlite3_open'
/usr/bin/ld: chisel_db.cpp:(.text+0x1048): undefined reference to `sqlite3_close'
/usr/bin/ld: chisel_db.cpp:(.text+0x107b): undefined reference to `sqlite3_backup_init'
/usr/bin/ld: chisel_db.cpp:(.text+0x1090): undefined reference to `sqlite3_backup_step'
/usr/bin/ld: chisel_db.cpp:(.text+0x1098): undefined reference to `sqlite3_backup_finish'
/usr/bin/ld: chisel_db.cpp:(.text+0x10a1): undefined reference to `sqlite3_errcode'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckCallocFromGroupFunc'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `vfs_timestamp'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `vfs_enable_multi_process_support'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `snps_mem_get_current_group'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `snps_mem_named_pop_group'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `vfs_feof'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `snpsSetSmallCaching'
/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `vfs_fflush'
[... cut ...]
**/usr/bin/ld: /home/cp1/tools/synopsys/vcs/vcs/Q-2020.03-SP2-7/linux64/lib/libvcsnew.so: undefined reference to `snps_mem_named_push_context'
collect2: error: ld returned 1 exit status**
make[2]: *** [Makefile:111: product_timestamp] Error 1
make[2]: Leaving directory '/home/cp1/xs-env.kunminghu.master/XiangShan/build/simv-compile'
Make exited with status 2
CPU time: 714.587 seconds to compile + 12.239 seconds to elab + 18.662 seconds to link
make[1]: *** [vcs.mk:85：/home/cp1/xs-env.kunminghu.master/XiangShan/build/simv] 错误 2
make[1]: 离开目录“/home/cp1/xs-env.kunminghu.master/XiangShan/difftest”
make: *** [Makefile:196：simv] 错误 2

cebarobot, 2024-01-12:
首先，您的机器上缺库，请通过 apt 安装它们。
First, your machine is lack of some libraries. Please use apt to install them.
```
apt-get install zlib1g-dev sqlite3 libsqlite3-dev 
```

其次，VCS 对 gcc 版本非常敏感。请先确保 VCS 在您的机器上可以正常运行，再尝试编译 VCS 版的香山仿真程序。
Second, VCS are indeed sensitive to the gcc version. Please make sure vcs can run properly first, then try to build vcs version of XiangShan emulator.

cebarobot, 2024-01-15:
请尝试将 `+dump-wave` 参数换成 `+dump-wave=vpd` 或 `+dump-wave=fsdb`。
Please try to replace `+dump-wave` with `+dump-wave=vpd` or `+dump-wave=fsdb`.

wyz-icer, 2024-01-15:
1. +dump-wave=vpd 可正常生成； 
2.  +dump-wave=fsdb   需要对vcs.mk 做如下调整
diff --git a/vcs.mk b/vcs.mk
index d3d04527..fb5d25f8 100644
--- a/vcs.mk
+++ b/vcs.mk
 # if fsdb is considered
-# CONSIDER_FSDB ?= 0
+CONSIDER_FSDB ?= 1
 ifeq ($(CONSIDER_FSDB),1)
 EXTRA = +define+CONSIDER_FSDB

cebarobot, 2024-01-15:
感谢您的反馈。您应该可以在 `make` 的时候直接添加 `CONSIDER_FSDB=1` 参数，这样不需要修改 Makefile。
Thank you for your feedback. You could add `CONSIDER_FSDB=1` when `make`, without modifying Makefile.

wyz-icer, 2024-01-16:
好的，谢谢！

closed 2024-01-16 by wyz-icer