OpenXiangShan/XiangShan #2661: 关于Case生成和nemu模拟器

changekkk, 2024-01-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior. Please include the commit hash and your modification as well.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (optional, if necessary):**
 - OS: [e.g. Ubuntu 20.04 LTS]
 - Compiler: [e.g. GCC 9.4.0]

**Additional context**
Add any other context about the problem here.

changekkk, 2024-01-20:
你好，想问下咱们目前使用的risc-v ture Case生成工具，以及验证比对模型nemu模拟器，是否支持risc-v的P扩展指令集？

cebarobot, 2024-01-20:
香山和 NEMU 目前均不支持 P 扩展，我们正在努力支持 V 扩展。

XiangShan and NEMU do not support P extension. We are working on V extension instead.

closed 2024-01-22 by poemonsense

changekkk, 2024-01-23:
目前香山的case生成，risc-v ture等工具，可以生成P扩展对指令类case吗

poemonsense, 2024-01-23:
香山目前没有提供指令生成器。如有相关需要可以去对应仓库询问