OpenXiangShan/XiangShan #2666: How to flush dirty datas to the Memory？

csgxiong, 2024-01-23:
Hi, I want to know about how to flush dirty datas to the Memory at XiangShan.
Are there some specific instructions or registers?

Thanks.

cebarobot, 2024-01-28:
Currently, XiangShan does not have a way to flush dirty data to memory directly.

RISC-V provides ISA extensions such as Zicbom, Zicbop, Zicboz to manage cache, but XiangShan does not yet support these extensions. Please check _[RISC-V Base Cache Management Operation ISA Extensions](https://drive.google.com/file/d/1jfzhNAk7viz4t2FLDZ5z4roA0LBggkfZ/view?usp=drive_link)_ or [RISC-V CMO TG](https://github.com/riscv/riscv-CMOs) for more information.

csgxiong, 2024-01-29:
I get it.
Thank you, cebarobot.