OpenXiangShan/XiangShan #2668: About getting the 128 bit floating-point function module.

whitecloudsun, 2024-01-23:
Hello，members of the Xiangshan team，I would like to ask a question as follows，

At present, I want to obtain four modules: floating-point 64 bit to floating-point 128 bit module, floating-point 128 bit FMA module, floating-point 128 bit addition module, floating-point 128 bit to floating-point 64 bit module. 

May I ask if it is possible to generate these four modules by modifying the initial scala code configuration? How do I need to proceed? Or can these be obtained separately by rewriting the 64 bit floating-point to 32-bit floating-point module, 64 bit floating-point addition module, and 64 bit floating-point FMA module generated under default configuration?

Looking forward to receiving your reply as soon as possible, thank you very much.

cebarobot, 2024-01-28:
At the moment, XiangShan does not support Q standard extension, specifically the Quad-Precision Floating-Point extension. I'm afraid that just modifying the Scala configuration won't allow you to obtain 128-bit modules.

Rewriting existing modules could be a choice, but that would entirely depend on your own efforts.

Tang-Haojin, 2024-03-15:
We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。