OpenXiangShan/XiangShan #2675: How Load Operations Perform Register Indirect Addressing

youzi27, 2024-01-26:
Hello. I would like to inquire about how XS handles the value of `rs1` during load operations (such as `LW`, `LBU`, etc.). 

For instance, is the address calculated as `rs1 + imm`, `rs1[7:0] + imm`, or in some other way?

When calculating the address for a load operation, is the value in the `rs1` register truncated in any way? Or is the complete value of `rs1`used for the calculation?

Thank you very much, I look forward to your response.

Tang-Haojin, 2024-01-27:
Please refer to the ISA manual. XiangShan would not truncate the operand in any way for a load instruction. If the address is insane, an exception will be raised.

youzi27, 2024-01-27:
> Please refer to the ISA manual. XiangShan would not truncate the operand in any way for a load instruction. If the address is insane, an exception will be raised.

Hi @Tang-Haojin . Thank you very much. I will close this issue and open a new one regarding the behavior I observed in XiangShan, which confuses me.

closed 2024-01-27 by youzi27

youzi27, 2024-01-31:
Hi @Tang-Haojin , I would like to ask how XiangShan handles situations where the immediate field of a load instruction exceeds the limits set by the instruction format?

Tang-Haojin, 2024-01-31:
This question is not related to XiangShan. Qualified compiler or programmer should know how to handle this.

For instance, say `ld a0, imm(a1)`, if `imm` is inadequate for you to address the offset, you can change the value in `a1`.

youzi27, 2024-01-31:
Thank you for your detailed explanation! If the imm field exceeds the immediate number range of the instruction format, should it trigger an illegal instruction exception?

Tang-Haojin, 2024-01-31:
Please read the ISA spec first. Such instruction does not exist.

[image]

youzi27, 2024-01-31:
Thank you, I will give an example.
For example, the immediate format for the c.fld instruction is 5 bits. If I execute c.fld fa1, 128(a1), 128 exceeds the maximum value that can be represented by a 5-bit range.
How does Xiangshan handle this situation?

eastonman, 2024-01-31:
Hello, your assembly will most likely be rejected by any RISC-V compiler.

Please stop using XiangShan issue as a replacement for GPT/Google, please.

Tang-Haojin, 2024-01-31:
You may try

`echo 'ld a0, 2047(a1)' | riscv64-linux-gnu-as -o a.out && riscv64-linux-gnu-objdump -S a.out`

and

`echo 'ld a0, 2048(a1)' | riscv64-linux-gnu-as -o a.out && riscv64-linux-gnu-objdump -S a.out`