OpenXiangShan/XiangShan #2676: Unexpected Memory Access Behavior in M-Mode Beyond Defined Address Space and PMP

youzi27, 2024-01-27:
**Describe the bug**
XiangShan in M-mode successfully accesses addresses far beyond the PMP range and even beyond the initialized address space defined for XS (e.g., 0xFFFF FFFF FFFF F000 ). This behavior differs from Spike, which correctly throws an exception for such accesses. 

**To Reproduce**
1. Configure XS in M-mode with a defined address space and PMP settings.
2. Attempt to access a memory address that is well beyond the PMP range even the defined address space.
3. Observe that XS allows the access without throwing an exception.
4. Perform the same test in Spike, which results in an exception for the same access.

**Expected behavior**
Although the specification states that 

> If no PMP entry matches an M-mode access, the access succeeds

attempting to load an address outside the defined memory range (e.g., an extremely large address) should raise an exception.

**Screenshots**
[image]

**Environment (optional, if necessary):**

- OS: Ubuntu 22.04.3 LTS
- Compiler: gcc 11.4.0

**Additional context**
pmpcfg0: 0x0f 
pmpaddr0: 0x0000000080002400

cebarobot, 2024-01-28:
Would you mind sharing a minimal test workload? This would greatly assist us in reproducing and addressing the issue.

youzi27, 2024-01-29:
> Would you mind sharing a minimal test workload? This would greatly assist us in reproducing and addressing the issue.

Hi @cebarobot . Thank you for your reply. To help you reproduce and address the issue effectively, I will prepare and share a minimal test workload. This will include the specific settings and steps that I used, which should provide a clear framework for identifying the problem. 

If you have any questions or insights, please feel free to contact me at any time. Thank you.

[testcase.zip](https://github.com/OpenXiangShan/XiangShan/files/14079473/testcase.zip)

good-circle, 2024-01-29:
The current physical address range for Xiangshan is 36 bits (0x00_0000_0000-0x0F_FFFF_FFFF), totaling 128G of address space. In the current PMA check rules, any address space greater than 0x00_8000_0000 is considered DDR space and supports read and write operations. 

For your own set PMP, although you may have set PMP permissions for a higher address space (e.g., addresses higher than 0x10_0000_0000), the Xiangshan processor currently only has a physical address of 36 bits. Maybe this is the cause of the phenomenon you're experiencing?

youzi27, 2024-01-29:
> The current physical address range for Xiangshan is 36 bits (0x00_0000_0000-0x0F_FFFF_FFFF), totaling 128G of address space. In the current PMA check rules, any address space greater than [...]

Hi @good-circle . Thank you for your response. However, the issue I'm raising doesn't completely align with your description. As outlined in the question, Xiangshan is capable of accessing some crazy addresses that far exceed 36 bits, such as 0xFFFF FFFF FFFF FFF0 (as seen in the testcase I provided).

good-circle, 2024-01-29:
Thank you for providing the testcase. We have successfully reproduced this phenomenon. The issue arises from the instruction `lw s1, 0(sp)`, where `sp=0xfffffffffffffff0`. In the loadunit's s0 stage, it needs to calculate the vaddr by adding 0 to 0xfffffffffffffff0. However, since the vaddr has a width of 39 bits, the high 1's are ignored. For `vaddr=39'h7f_ffff_fff0`, when converted to a 36-bit physical address, it becomes `paddr=36'hf_ffff_fff0`. As the address 36'hf_ffff_fff0 actually falls within the DDR range defined by the processor, no access fault is reported.

The reason for this issue is the disregard of the high-order information in the physical address (in M-Mode, this corresponds to the high-order information of the virtual address). When we enable virtual-to-physical address translation, we consider this issue. Currently, our physical address has a width of 36 bits. During TLB page table walk, if the query reveals that the PPN high-order bits are not 0, an access fault will be reported. However, we have not yet addressed the scenario where virtual memory is not enabled.

youzi27, 2024-01-30:
Thanks a lot for getting back to me so quickly. I've noticed the same issue, which led me to disable virtual address translation in the test case. Despite that, the issue I mentioned still occurs, which seems to match up with your comment about ’However, we have not yet addressed the scenario where virtual memory is not enabled‘.

Thank you for verifying this issue.

poemonsense, 2024-01-30:
> The reason for this issue is the disregard of the high-order information in the physical address (in M-Mode, this corresponds to the high-order information of the virtual address).

Duplicated and already known in https://github.com/OpenXiangShan/XiangShan-internal/issues/1

youzi27, 2024-01-30:
> > The reason for this issue is the disregard of the high-order information in the physical address (in M-Mode, this corresponds to the high-order information of the virtual address).
> 
> Duplicated and already known in https://github.com/OpenXiangShan/XiangShan-internal/issues/1

Hi @poemonsense. Thank you for sharing your insights regarding the handling of high-order information in physical addresses. Unfortunately, I'm unable to view the link you provided as it seems to be restricted to XiangShan's internal members.

From what you've mentioned, it appears the issue is considered primarily within the context of virtual address translation being active in M-mode. However, my observations suggest that this issue isn't confined to scenarios where virtual address translation is enabled.

In fact, I've noticed that the problem also arises in situations where virtual memory isn't utilized, particularly in M-mode. The XiangShan seems to mishandle addresses well outside the defined memory range, even with virtual memory turned off. This goes against the specifications, where such out-of-range accesses should raise an exception. Yet, the XiangShan doesn't seem to recognize these addresses as invalid or illegal.

Thank you again for your attention to this issue.

cebarobot, 2024-01-30:
> Great analysis! By the way, I would like to ask, does XiangShan support SV48 address translation?

No, XiangShan does not support SV48 at present.

closed 2024-02-26 by Lemover