OpenXiangShan/XiangShan #2699: DCacheTest failed

SpinEch0, 2024-02-19:
I write a DCache Module Test case. It failed with below message:
```
========== in test ==========
Assertion failed
    at WritebackQueue.scala:532 assert(!RegNext(!s_data_merge && !s_data_override))
```
I am sure s_data_merge and s_data_override are not both 0.
Maybe RegNext should be removed or with an initial value ?

```
assert(!(!s_data_merge && !s_data_override))
```

SpinEch0, 2024-02-19:
same problem: 
```
Assertion failed
    at MissQueue.scala:1010 assert(RegNext(PopCount(secondary_ready_vec) <= 1.U))
Assertion failed
    at MissQueue.scala:683 assert(RegNext(PopCount(Seq(io.primary_ready, io.secondary_ready, io.secondary_reject)) <= 1.U))

```

cebarobot, 2024-02-19:
Please provide more information, such as the details of your testcase.

By the way, I kindly remind you that you SHOULDN'T remove the issue template. Please always fill the issue template when opening an issue.

SpinEch0, 2024-02-19:
PrefetcherMonitor.scala:152 assert(depth =/= 0.U, "depth should not be zero")  depth is not initiated

closed 2024-02-23 by SpinEch0