OpenXiangShan/XiangShan #2703: outstanding Capability of nanhu v2 uncache access

menglinhan, 2024-02-21:
您好，我在trace 南湖v2的代码时，发现uncahceable的对外访问（uncache模块下），有一个state的状态机会限制core对外uncahce读写的能力在1，a通道的valid需要d通道有valid（也就是前一笔访问的response）后才会发新的访问。XSTille的uncache读写outstanding能力为1，请问是这样吗？

poemonsense, 2024-02-22:
1. 鉴于您已经连续提出多个删除了issue模板的issue，我们提醒您：请不要删除issue模板中的内容，并按照模板中的规范进行提问，以方便其他人的搜索和知识的传播

2. 我们在issue模板中提醒，请阅读香山文档，您的问题在文档中已经有回答：https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/soc/#axi-interface-axi

> * 64-bit AXI Peripheral Port: for CPU to access peripherals (CPU is the master)
> Address space: peripheral slaves space (0x0000_0000 to 0x7fff_ffff)
> AXI ID width: 2-bit
> Number of outstanding: 1 for instruction and 1 for data for each core

closed 2024-02-22 by poemonsense

menglinhan, 2024-02-22:
了解了，多谢，后面我会注意的